pvlib/pvlib-python #716: Docstring for pvsystem.calcparams_xxx are inconsistent with function code

cwhanse, 2019-05-09:
Docstrings for `pvsystem.calcparams_desoto`, `pvsystem.calcparams_cec` and `pvsystem.calcparams_pvsyst` list outputs as `photocurrent` etc. but code uses different terms, e.g., `IL`.  Using the long variable names inside the functions would clarify the code. 

Might want to consider returning dict rather than tuple.

wholmgren, 2019-05-09:
Ok with me to change the code to use long names. But I don't think the long/short inconsistency between the docstring/API and internal variable names is necessarily a problem. Our [Contributing docs](https://pvlib-python.readthedocs.io/en/stable/contributing.html#code-style) say:

> pvlib python uses a mix of full and abbreviated variable names. See Variables and Symbols. We could be better about consistency. Prefer full names for new contributions. This is especially important for the API. Abbreviations can be used within a function to improve the readability of formulae.

cwhanse, 2019-05-09:
Maybe adding a comment line with the return statement will close the bit of gap between the docstring and the variables used in the code.  I think it is helpful to use the common symbols as variable names. I don't think including the symbols in the docstring would improve matters; seems like step back from the Contributing goal of full names.

pratham15541, 2025-02-18:
Can i do like
Updating the docstrings for the mentioned functions to match the variable names used in the code.
Modifying the functions to use long variable names for clarity.
Consider changing the return type of the functions from a tuple to a dictionary to improve readability.

closed 2025-03-31 by cwhanse