pvlib/pvlib-python #729: Remove Python 2 vestiges

markcampanelli, 2019-05-20:
With the merging of #728, we can now consider dropping Python2-specific code and workarounds.

Some candidates are:
- https://github.com/pvlib/pvlib-python/blob/master/pvlib/atmosphere.py#L214
- Uses of `from __future__ import ...`
- Other places where Python 2 vs. 3 considerations may have "[complected](https://www.youtube.com/watch?v=rI8tNMsozo0)" the code.

alexandermorgan, 2019-07-21:
@markcampanelli your list above is good, but the first point seems to be a separate issue, since it's not clear to me that the current behavior returns the correct number at all, in py2 or py3. At least that's the way the comment on line 214 makes it sound.
I've made a PR that addresses the __future__ imports. They're mostly importing division, but also one print_statement for the versioneer. Should this versioneer __future__ import be left alone since it's a separate entity? has_python2 checks are also removed from conftest.py and test_conftest.py.
Do you agree with the plan of closing this issue after merging my PR #757, and then making a new issue to address your first point above?

markcampanelli, 2019-07-22:
I think that first item was simply what I discovered from just searching the code for Python 2 references. I think it's fine to address that in a separate issue/PR. I don't see a problem with also removing the versioneer future presently if it is not needed.

I think you should also search the code for "python 2" and variants. For example, this identifies import checks that are no longer needed, such as https://github.com/pvlib/pvlib-python/blob/b61e149226004d4e5ba0e49da61d180173372fee/pvlib/iotools/epw.py#L9

alexandermorgan, 2019-07-22:
Thanks, I've sorted out all the py2-compatible imports. Also, after reading the commit where the comment about python 2 was added in atmosphere.py, I see that we don't need to make a new issue for that. We can just remove the comment because the solution previously found seems to work fine in py2 and py3, and looks a little better than the approach that was only py3 compatible, so no problems there after all.

PR #757 (2019-07-26): remove python 2 lines
 - [X] Closes #729
 - [X] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is sufficient. Passes LGTM and SticklerCI checks [... cut]
closed 2019-07-26 by wholmgren