pvlib/pvlib-python #736: v0.6.4

cwhanse, 2019-06-11:
Given other demands and the anticipated scope of 0.7.0, we should have at least one more minor release before 0.7.0.  Opening this issue to discuss what goes in 0.6.4, beyond issues/PRs with the milestone label.

#718 ?  Will provide one way to close #587 and closes #511.  #708 also?  belongs in the same family.

Other issues to close with v0.6.4?

mikofski, 2019-06-12:
OK, sounds good. I'll put some effort into closing #708 and making sure it will work with #718. Any guidance would be appreciated.

cwhanse, 2019-06-13:
Happy to help with #708 what issues are still outstanding?

wholmgren, 2019-06-16:
Looks like a 0.6.4 tag was accidentally created and pushed so a pypi release was triggered. Does it need to be deleted? If so I will find time to do it next week. 

https://pypi.org/project/pvlib/

cwhanse, 2019-06-17:
Please delete. I confused a tag with a milestone, wanted the latter.

wholmgren, 2019-06-18:
I deleted 0.6.4 from pypi. The next release must be 0.6.5 or greater.

wholmgren, 2019-06-18:
On the merits of this... I suggest that we move ahead with 0.7 rather than make another 0.6.x release. We can reduce the scope of 0.7.0 if we want to. The next release is the first that is python 3 only, which also argues in favor of a 0.7 release rather than 0.6.5. It would be possible to create an 0.6 maintenance branch for parallel development but that is not something I'm willing to put time into.

cwhanse, 2019-06-18:
I'm OK reducing the scope if we're ready to drop python 2 coverage. I don't think a parallel branch is worth the effort until we hear from someone that they need it.

We planned some API-breaking changes e.g. #682 for 0.7. Do we want to keep those in scope, or push them down the calendar?

wholmgren, 2019-06-18:
I suggest keeping #682 in scope. From a quick look, I think it only needs minor updates to the deprecation wrappers to indicate that user code will break starting with 0.8.

cwhanse, 2019-06-25:
We are skipping v0.6.X and moving to v0.7.0 as the next release

closed 2019-06-25 by cwhanse