pvlib/pvlib-python #739: port MATLAB PVLIB's Boyle/Coello soiling model to python

wholmgren, 2019-06-24:
https://github.com/sandialabs/MATLAB_PV_LIB/pull/26 

help wanted...

alexandermorgan, 2019-07-19:
I would like to participate in this issue. Is there a preference as to which model to implement first, i.e. Kimber Dynamic or Boyle/Coello? I could start by doing a pandas rewrite of the code @mikofski shared in issue #669. Mark's code was very clear and in Python, so maybe it's easiest to start with the Kimber Dynamic method.

nappaillav, 2020-01-08:
Hello 
As per the suggestion of @cwhanse, I have started working on, porting of the code from Matlab to python. I have taken reference from `pvl_soiling_hsu.m` code. 
I have few doubts like should I reproduce the whole Matlab code (both pvl_soiling_hsu and depo_velocity or just the depo_velocity).

DanRiley, 2020-01-08:
@nappaillav If you have questions when porting any part of the Humboldt State University (HSU) code, feel free to contact me. I worked on developing that code from the code that was sent to me by the paper authors. 

I recommend reproducing the entirety of the MATLAB code. The internal function depo_velocity is used by the variable deposition form of the model, but requires everything in pvl_soiling_hsu.m in order to correctly model the soil impact.

cwhanse, 2020-01-08:
I think the expected behavior should be soiling_ratio = f(inputs), as `pvl_soiling_hsu.m` does. The MATLAB input `ModelType` selects the soiling deposition model from 3 options (option 1 uses delo_velocity helper function), I'd default to `ModelType=2`.

FYI: the paper authors requested us to name the function with HSU rather than a last name. It would be nice that this function become known as the Humboldt State University soiling model to honor their request.

nappaillav, 2020-01-08:
I'm trying to handle the argparse in python similar to matlab. 
Else I have a complete code ported in python. Apart from that I'm yet to test the functions.
Hence It would be great if you can provide me with some test data which I can use to see if the python code reproduces the same values as the HSU code.

cwhanse, 2020-01-08:
I wouldn't work too hard to imitate the MATLAB argument parsing. You can take defaults from the MATLAB function's Optional inputs. I wouldn't port the validation rules; rather, describe the value restrictions in the docstring.

@DanRiley do you have a few sets of test data?

PR #850 (2020-01-17): add Boyle/Coello (Humboldt State Univ) soiling model

 
 - [X] Closes #739
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [X] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [X] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` : [... cut]
closed 2020-01-17 by cwhanse