pvlib/pvlib-python #749: hour_angle

cesar3158, 2019-07-10:
Hi!!

I'm trying to compute the hour angle for a certain local time using the pvlib's function "**pvl.solarposition.hour_angle()**".

From a GPS time in seconds of week I getting the date using following code:
**datetimeformat =  ('%Y-%m-%d %H:%M:%S')
leapsecond = 37
epoch = datetime.datetime.strptime ( "1980-01-06 00:00:00" , datetimeformat)
decorrido = datetime.timedelta( days = (2024 * 7 ), seconds = t_gps + leapsecond)
final_time = datetime.datetime.strftime(epoch + decorrido, datetimeformat)
print(final_time)
print(type(final_time))**
The printed result and the variable type are:
**2018-10-22 14:22:05
<class 'str'>**

The time printed (**14:22:05**) is already in UTC time because in Brazil the local time is **12:22:05**.

The result obtained to the variable "final time" will be used to calculate the hour_angle using the following code:

**hour_angle = pvl.solarposition.hour_angle(final_time,  lat_long[0][1],  eq_time)
print(hour_angle)**

When I try to run the calculation for "hour_angle" the python returns the following error:
**naive_times = times.tz_localize(None)  # naive but still localized
AttributeError: 'str' object has no attribute 'tz_localize'**

I would like to understand how I should proceed to calculate the value for hour_angle, informing to the function that the time for "**final time**" value is already in UTC Date Time.
To solve this issue I would have to turn the result obtained to "**final time**" to local time and then, reapply the conversion to UTC Data Time and this result be used to hour_angle calculation?

Thank you for help me!

Regards.

mikofski, 2019-07-10:
Hi @cesar3158, thanks for using pvlib and for your question. Please see the documentation for [`hour_angle`](https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.solarposition.hour_angle.html#pvlib.solarposition.hour_angle). Also consider posting this usage question on stack overflow with the tag [pvlib] or on the pvlib google group for support as described in the [pvlib documentation](https://pvlib-python.readthedocs.io/en/latest/package_overview.html#getting-support/). Please don't hesitate to ask more questions. Thanks!

cwhanse, 2019-07-10:
@cesar3158 You'll need to localize `decorrido` to the time zone of the longitude, and pass `decorrido` instead of the string `final_time`.  If you can post reproducible code on stackoverflow or the google group I can look into it a bit more.

closed 2019-07-10 by wholmgren