pvlib/pvlib-python #751: Implement IEC61853 IAM calculations

adriesse, 2019-07-11:
The standard gives the option of using interpolation of measured data, or a model which has the single parameter `a_r`.  Both should be implemented.

cwhanse, 2019-07-11:
The [Martin and Ruiz model](https://github.com/sandialabs/MATLAB_PV_LIB/blob/master/pvl_iam_martinruiz.m) is in PVLib for MATLAB. In a later paper Martin and Ruiz extended the model to estimate reflection of diffuse irradiance, also in PVLib for MATLAB [here](https://github.com/sandialabs/MATLAB_PV_LIB/blob/master/pvl_iam_martinruiz_components.m)  I think both versions are useful, the component version shows up in various bifacial irradiance models.

cwhanse, 2019-07-11:
#679 and #680 are relevant

PR #752 (2019-07-26): Implement IEC 61853 IAM calculations
 - [x] Closes  #751 
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is sufficient. Passes LGTM and  [... cut]
closed 2019-07-26 by cwhanse