pvlib/pvlib-python #754: Rounding issue in clearsky.py _linearly_scale method

alexandermorgan, 2019-07-16:
**Describe the bug**
In the _linearly_scale method, there are two logic statements (lines 316 and 321) that first round values to one decimal place, then compare if <= .5. Since they're being rounded first, values like .54 would become .5, then the check <= .5 would actually be true. Also, from a performance standpoint, there is no reason to round first, we could instead just do the comparison immediately. If you actually want to check if values are < .55, then write that explicitly and modify the comment on line 314. Or keep <= .5 and the leave the comment as is, but either way, there's no need to round and compare to a float. I can make a pull request for either approach, please just let me know what's preferred.

**To Reproduce**
```
>>> import numpy as np
>>> arr = np.array([.49, .50, .54, .55, .56])
>>> np.around(arr, 1) <= .5  # current behavior
array([ True,  True,  True, False, False])
>>> arr <= .5  # intended behavior and also much faster
array([ True,  True, False, False, False])
>>> arr < .55  # equivalent to current behavior but much faster
array([ True,  True,  True, False, False])

```
**Expected behavior**
I would have expected values like .54 to raise an error in this if block, but they don't.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions:**
 - ``pvlib.__version__``: 0.6.3
 - python: 3.5.2

**Additional context**
Removing the unnecessary rounding operation reduces the runtime of these logic checks by about 60%.

wholmgren, 2019-07-16:
@mikofski you added this code in #264. Thoughts?

mikofski, 2019-07-16:
No, I can't remember what my thinking was here. I believe @alexandermorgan is correct, and the call to `np.around()` is unnecessary. Thanks for the catch!

alexandermorgan, 2019-07-17:
Ok, I'll make a PR that removes the np.around calls and uses <= .5 comparisons (the intended, but not current, behavior).

alexandermorgan, 2019-07-18:
@mikofski I made a patch with these changes, but it fails pvlib/test/test_clearsky.py -> test_likne_turbidity_corners(). It fails because the current approach basically rounds anything below .55 to .5. So one of the test cases has these values:
outputmin = 0
outputmatrix = -0.5000000000000568
So in line 320 of clearsky:
        if np.around(output - outputmatrix, 1) <= .5:
This evaluates to true, because it rounds the value that's slightly above .5, down to .5 first, before checking if it's <= .5. Do you want to preserve this behavior? If so, then I'll just change the comparison to <= .55. I think I should change the test though, to assert that it raises an error with this value. If you're worried about floating point inaccuracies, I could make the comparison <= .50000001 or something like that.

mikofski, 2019-07-31:
Aha, yes this explains why I had originally rounded, because of the issue in comparing floats.

    # If the index is still out of bounds after rounding, raise an error.
    # 0.500001 is used in comparisons instead of 0.5 to allow for a small
    # margin of error which can occur when dealing with floating point numbers.

Good catch, I agree with good

PR #755 (2019-08-01): fix rounding issue in linke turbidity lookup function
pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with your code at any stage of completion.

The following items must be addressed before the code can be merged. Please don't hesitate to ask for help if you're unsure of how to accomplish any of the items below:

 - [X] Closes #754 
 - [X] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.i [... cut]
commit 0be834c1 (2019-08-01): fix rounding issue in linke turbidity lookup function (#755)

* Resolve merge conflicts.

The upstream what's new file had been updated.

* Copy whatsnew file from upstream.

* Remove unnecessary np.around().

Instead of rounding to one decimal place, then comparing to .5,
just compare to .55 right away which preserves the same behavior as
before. If you want to stay true to the comment on line 314, use
<= .5 in comparison instead as before, but still drop the np.around().

* Use <= .5 as comparisons in _linearly_scale.

* Correct previous commit, use <= .5.

* Update what's new file.

Closes  [... cut]
closed 2019-08-01 by cwhanse