pvlib/pvlib-python #756: Isotropic Sky Diffuse geometry

JPalakapillyKWH, 2019-07-18:
**Describe the bug**
The traditional conversion factor ((1 + cos(tilt)/2) used in the isotropic sky model to calculate sky diffuse irradiance is geometrically inaccurate. The implementation of this model is in pvlib.irradiance.isotropic()

The original model that Hottel and Woertz developed is only a 2-dimensional consideration. The geometry argument is best put forward in https://doi.org/10.1016/j.renene.2011.06.042

There are two suggested improvements. One is the linear expression put forward by Tian et al. The other is the diffuse tilt factor (DTF) developed in the above paper. Both are more geometrically sound than the original model by Hottel and Woertz.

cwhanse, 2019-07-22:
@JPalakapillyKWH still working through the papers. There might be a conflation of DHI received on the ground (in W/m2) with incoming sky luminance (in W/m2/steradian). The diffuse tilt factor appears to account for both sky diffuse and ground reflected irradiance - is that the way you also read it?

JPalakapillyKWH, 2019-07-22:
Yep, the DTF does account for both. I am exclusively looking at the term that captures the sky diffuse irradiance (1 + cos^2(tilt))/2 which I thought at the time was more geometrically accurate than the current conversion factor. However, after looking at the math more closely, I believe the integration done in the Rakovec paper is also flawed.

Geometrically, I think the correct thing to do is to integrate the projection of the irradiance onto the receiving surface as done in the Rakovec paper. But this should be done over the visible part of the hemisphere as shown in Fig. 2 of the Rakovec paper. If you look at the limits of the integration in the Rakovec paper, you can see that the implied solid angle doesn't correctly capture the visible part of the hemisphere. (The limits on the integration of the zenith angle should not be from 0 to pi/2 - tau for half of the hemisphere)

Unfortunately, my calculus/geometry knowledge is limiting me from finding a closed form solution to the integration that I think is correct. But I did a numerical integration and the results are identical to the original conversion factor of (1+cos(tilt))/2.

I still believe (1+cos(tilt))/2 is not the correct sky view factor (SVF) for the geometric reasons outlined in the Rakovec paper. Tian's linear model is a geometrically accurate SVF. However the sky view factor doesn't take into account the relative angles between incident irradiance and the receiving surface so it is irrelevant. The proper conversion factor between DHI and the sky diffuse irradiance for a tilted plane should consider the relative angles and does seem to be (1+cos(tilt))/2 .

All of this to say that I think the conversion factor in PVlib is actually right (although maybe for the wrong reasons). This issue I think should be closed :)

cwhanse, 2019-07-23:
@JPalakapillyKWH I agree that in this case, (1+cos(tilt))/2 is not a view factor in the usual meaning. The integration over the visible portion of the sky is shown in [2], but that calculation is also accounting for reflections. To get the (1+cos(tilt))/2 result set b_0 to 0.  
[1] D. J. Wright, "Closed-Form, Analytic Solution for Diffuse Irradiance on a Tilted, Photovoltaic Collector," in IEEE Journal of Photovoltaics, vol. 9, no. 2, pp. 391-396, March 2019.
doi: 10.1109/JPHOTOV.2018.2888825

closed 2019-07-23 by cwhanse

JPalakapillyKWH, 2019-07-23:
Ah thank you so much for linking that paper, Cliff. I can have some closure after seeing an analytic derivation.