pvlib/pvlib-python #759: read ISD files

cwhanse, 2019-08-01:
Add functions to read files from the [Integrated Surface Database (ISD)](https://www.ncdc.noaa.gov/isd) .  

One possible approach is to port pvl_getISDdata.m and pvl_readISH.m from [PVLib Matlab](https://github.com/sandialabs/MATLAB_PV_LIB). 

A parser is in rOpenSci at [CRAN](https://github.com/cran/isdparser). 

A older python [reader](https://github.com/haydenth/ish_parser) could be used, but it is questionable if the code is being maintained. It uses many classes and may not fit well with pvlib's style.

wholmgren, 2019-08-01:
Porting from PVLib Matlab would probably be fastest. It looks like NOAA now discourages the ftp access route and recommends the http archive with proper csv files, so the port might end up being fairly independent.

What do you think about putting the code in [siphon](https://github.com/unidata/siphon) instead? The ISD is more general purpose than most of the datasets accessible in `pvlib.iotools`. 

I agree with your assessment of the older `ish_parser`.

cwhanse, 2019-08-01:
> What do you think about putting the code in siphon instead? The ISD is more general purpose than most of the datasets accessible in pvlib.iotools.

Certainly. Or [metpy](https://unidata.github.io/MetPy/latest/api/generated/metpy.io.html#module-metpy.io)

I'd rather import the code from a maintained library than maintain it ourselves.

mikofski, 2019-08-07:
I would also like to see a reader for [NOAA NCDC GSOD](https://data.noaa.gov/dataset/dataset/global-surface-summary-of-the-day-gsod), especially for rainfall (see [ftp](ftp://ftp.ncdc.noaa.gov/pub/data/gsod/) see #668

nappaillav, 2020-01-20:
I would like to work on this issue.

So here is my plan to start :
I would first go through the matlab codes 1) `pvl_getISDdata.m` and 2) `pvl_readISH.m`
also there is a parser in **CRAN's** repository. 

So what would you recommend me to do? if there is any other information related to this issue please do let me know :)

cwhanse, 2020-01-20:
@nappaillav from the discussion above, I think a good first step is asking if siphon or metpy is interested to implement an ISD reader.

nappaillav, 2020-01-21:
@cwhanse have look into this [eeweather](https://eeweather.readthedocs.io/en/latest/) which supports Integrated Surface Database (ISD).

cwhanse, 2020-01-21:
> @cwhanse have look into this eeweather which supports Integrated Surface Database (ISD).

That's a neat package. I hadn't found that, thanks for posting it. I would have to use eeweather with pvlib for a bit, to figure out if there's value in new pvlib code as an interface.

Plusses: 
- forked by volttron so eeweather could get long-term use.
- assists with finding stations by lat/lon

Minuses: 
- long list of dependencies, pretty heavy for pvlib to import
- few developers but forked by volttron

Any response from siphon or metpy?

nappaillav, 2020-01-21:
@cwhanse I will rise a issue in both and let you know soon siphon and metpy

nappaillav, 2020-02-14:
I had raised a feature request in [Siphon](https://github.com/Unidata/siphon/issues/299) and [metpy](https://github.com/Unidata/MetPy/issues/1315) long ago, for which I got a response.  The author feels that ISD reader function suits Siphon more than metpy. 

But I'm not sure by when they will implement the function.

cwhanse, 2020-02-14:
Seems like the responses are "PR welcome" and put `get_` functions into `metpy` and `read_` functions into `siphon`.

Having looked into it, I'm -1 on eeweather. Although that package has nice functions, it primarily serves a particular workflow, so maintaining a general interest function to read ISD is not likely a priority.

@wholmgren do you see a downside for pvlib users if the ISD functions live in siphon/metpy rather than native in pvlib?

wholmgren, 2020-02-14:
No, I don't see a downside. You'd have to install another package or two, but their base requirements are not difficult to manage. @nappaillav I'm sure that the siphon/metpy maintainers would welcome a pull request.

AdamRJensen, 2023-05-15:
@wholmgren and @cwhanse Can this issue be closed (based on the above discussion it seems so)?

cwhanse, 2023-05-15:
@AdamRJensen I think so. We can always re-open.

closed 2023-05-15 by cwhanse