pvlib/pvlib-python #766: loading bifacial module

toddkarin, 2019-08-16:
The __init__.py should have a line
```python
from pvlib import bifacial
```
to import the bifacial module. Otherwise it can't be used!

markcampanelli, 2019-08-16:
While I'm not sure if there are criteria for adding this line (e.g., stability of the automatically imported module), there is definitely a work around in the meantime. This works at a terminal:
```terminal
python -c "import pvlib.bifacial as bifacial; print(type(bifacial)); print(bifacial.PVFactorsReportBuilder())"
```

toddkarin, 2019-08-17:
Yes, that's a workaround -- it just took me a moment to figure this out since the pvlib documentation says there is a bifacial module, but it is not loaded by default (unlike all the other modules). Someone with less python experience might not be able to figure this out.

wholmgren, 2019-11-26:
Thanks @toddkarin and sorry for missing this until now.

PR #826 (2019-11-27): import bifacial module in __init__.py

 - [x] Closes #766 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with ` [... cut]
closed 2019-11-27 by wholmgren