pvlib/pvlib-python #768: Avoid replacing so many characters with `_` in `_parse_raw_sam_df()`

Peque, 2019-08-23:
As [suggested by @wholmgren](https://github.com/pvlib/pvlib-python/pull/753#issuecomment-513292390).

I too think it would be more convenient. For example, it would allow for easier distinction between manufacturer and module number.

Peque, 2019-09-22:
@wholmgren @cwhanse Since we already broke the API by changing the module and inverter databases (i.e.: modules/inverters name need to be updated to work with 0.7), would you be okay to introduce this change as well?

Asking before spending time preparing the PR. :blush:

Peque, 2019-09-22:
Just adding [another vote](https://github.com/pvlib/pvlib-python/pull/753#issuecomment-513456822) in favor of this change:

> @adriesse 
> 
> I've never use the tab completion and find the database more convenient in its non-transposed form.

cwhanse, 2019-09-26:
I agree with making changes to the module dataframe keys. I think our goal is to remain as consistent as possible with the [source](https://github.com/NREL/SAM/blob/develop/deploy/libraries/CEC%20Modules.csv). We can propose changes to the source, if that's the best course. The SAM developers agree with the overall goal to have one name for a module that is common among both codes.

Peque, 2019-09-26:
@cwhanse @wholmgren I would propose to have a separator between the manufacturer name and the model name, either a `,` (separate columns in the CSV) or some other character. It used to be the case (a `:` was used), but it seems not anymore. :confused: 

I opened this issue, in case you want to upvote/comment: https://github.com/NREL/SAM/issues/227

I may wait for that to be resolved before implementing this change (feeling lazy, in case I have to change the names 2 times... :joy:)

cwhanse, 2019-11-05:
@Peque looking for closure on v0.7 release. Would you prefer to defer work in pvlib on this issue, pending any changes in SAM?

Peque, 2019-11-06:
@cwhanse Thanks for the ping.

I would prefer to defer work pending changes in SAM, if you do not mind.

cwhanse, 2019-11-06:
@Peque don't mind at all.

Peque, 2020-06-13:
Maybe related (creating a separate quick-to-load Python package for parts): https://groups.google.com/forum/#!topic/pvlib-python/Sqe84M2C6RY