pvlib/pvlib-python #770: Raise an exception if no parameter is provided to `retrieve_sam()`

Peque, 2019-08-23:
Instead of opening a dialog, which feels weird for a library.

https://github.com/pvlib/pvlib-python/blob/0a14b273517a082603cc157faa88a5ab0ac4cac9/pvlib/pvsystem.py#L1783-L1788

wholmgren, 2019-08-23:
I agree. We could deprecate but I think we should just go ahead with the breaking change. Does anyone reading this rely on this behavior?

mikofski, 2019-08-23:
I'm okay with breaking change here because a dialog is a user interaction vs a programming interface, so change doesn't break the API, just potential user flow. Either way keep it or not, although lack of testing is probably argument for dumping it

adriesse, 2019-08-24:
I have a vague memory of the dialog not working for me when I inadvertently tried to use it a few months ago.  Prefer exception to dialog.

commit b4238beb (2019-08-28): Raise if no parameters provided to retrieve_sam() (#770) (#775)

Close #770 via #775