pvlib/pvlib-python #771: add sphinx documentation build to CI

wholmgren, 2019-08-23:
A handful of recent PRs have broken the documentation build. We should add the documentation build to the azure pipelines CI configuration and fail it if the documentation errors.

Peque, 2019-08-28:
Maybe you want to have a look at [Autobuild Documentation for Pull Requests](https://blog.readthedocs.com/building-docs-for-pull-requests/), since you are already using Read the Docs and GitHub.

Peque, 2019-08-28:
Yeah, it is in beta. [You can currently email them](https://blog.readthedocs.com/building-docs-for-pull-requests/#getting-started) to enable the feature. Just wanted to link it here in case you want to try it early or in case it comes out of beta before anybody fixes this issue.

They will probably be faced with a large increase in compute needs, yeah, but I hope they can compensate that with paying subscriptions, as this was a largely requested feature for private repositories as well. :blush:

wholmgren, 2019-08-28:
Thanks, I emailed them.

wholmgren, 2019-08-29:
RTD added us the the beta program and I modified our RTD webhook to support PR doc builds. Docs for existing PRs will be built after a new commit is pushed. For example: https://external-builds.readthedocs.io/html/pvlib-python/773/index.html

Thanks again for pointing out this feature!

CameronTStark, 2020-02-22:
I've been thinking about the gallery contributions and wondering how they can be tested within the CI. I'd like to make sure that these contributions  are kept up to date with API changes so newbies don't have to worry about broken examples. 

I'm just starting to explore this and might open a separate issue to address but I at least wanted to begin to explore the idea here.

kanderso-nrel, 2020-02-22:
It's not a full test in that it doesn't check for expected output, but sphinx-gallery has a

CameronTStark, 2020-02-23:
Thank you for that @kanderso-nrel. An error will be raised which should fit our needs with the CI.

mikofski, 2020-02-23:
Sphinx had built in [doctests](http://www.sphinx-doc.org/en/master/usage/extensions/doctest.html). Enable it like any other extension by adding `sphinx.ext.doctest` to the `conf.py`. SciPy uses this so that all examples are legit. It doesn't run on the CI, but the docs won't build on RTD if any examples fail.

PR #909 (2020-02-26): BLD: build docs on Azure Pipelines

 - [x] Closes #771
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [x] Pull request is nearly [... cut]
closed 2020-02-26 by wholmgren