pvlib/pvlib-python #779: change readthedocs link to point to stable

mikofski, 2019-09-11:
### Problem
The links on the GitHub page at the top and in the `README` both refer to _latest_ which currently is not rendering some of the IPython and figures in [Intro Examples](https://pvlib-python.readthedocs.io/en/latest/introexamples.html).

### GitHub top
[image]

### README
[image]

### Latest Intro Examples
[image]
[image]

### Pypi
However, the link to the docs on PyPI do point to _stable_, which seems to be rendering correctly.

[image]

### Proposal
Should we make all of the links point to _stable_? I can also see an advantage to having the GitHub links point to _latest_ assuming that GitHub readers are developers, and okay with seeing the bleeding edge which might be broken.

Sorry if this is already addressed in the documentation or in a prior discussion somewhere. I didn't see anything related in [contributing or documentation](https://pvlib-python.readthedocs.io/en/latest/contributing.html#documentation) sections.

wholmgren, 2019-09-11:
I did change https://pvlib-python.readthedocs.io to point to stable a while ago, but I forgot that the github links are specifically to latest. I lean towards making all of them point to the root url and letting the RTD configuration specify the pointing, but I see your point about maybe the Github links should go to latest.

mikofski, 2019-09-11:
Sorry, I was suggesting the opposite. They already point to latest.
>but I see your point about maybe the Github links should go to latest.

I meant thanks if there was a good reason for them to point to latest, I would be open minded.

>I lean towards making all of them point to the root url and letting the RTD configuration specify the pointing

I agree with this

wholmgren, 2019-09-11:
I just changed the link at the top of the Github repo page to https://pvlib-python.readthedocs.io, so RTD will redirect that as specified (currently to stable). 

I then went to edit the README and thought better of it. All of the other badges reflect the current state of master, so I think the RTD badge should continue to point to latest. Perhaps a second badge could be added, but seems unnecessary. I don't think it actually needs to be consistent with the top of page link -- likely different kinds of users clicking on them.

I guess the `setup.py` project description (what shows up on PyPI) could explicitly direct to stable, but that seems unnecessary for as long as RTD redirects to stable.

closed 2019-09-11 by mikofski