pvlib/pvlib-python #782: reading bug on '*_TMY3.epw' file

tylunel, 2019-09-30:
**Describe the bug**
When a tmy3 file formatted in epw is read, the precipitable_water is not converted from mm to cm, and it results in diverging results for spectral mismatch factor (like negative factor for instance), then resulting in negative 'effective_irradiance' results

**To Reproduce**
Steps to reproduce the behavior:
1. Create a ChainModel object with a .epw weather file, coming from TMY3 data and imported with 'pvlib.iotools.epw.read_epw()'
2. Run the model
3. Check for inconsistencies in effective_irradiance with 'chain1.effective_irradiance.loc[chain1.effective_irradiance<0]'

or 
1. Run the code attached
    
**Expected behavior**
All the effective_irradiance should be positive or null.

[bug_report_read_epw.txt](https://github.com/pvlib/pvlib-python/files/3672185/bug_report_read_epw.txt)

**Versions:**
 - ``pvlib.__version__``: 0.6.3+24.gbbb4580.dirty
 - ``pandas.__version__``: 0.23.4
 - python: 3.7.1

cwhanse, 2019-09-30:
@roelloonen would appreciate your input here. To me, it appears that EnergyPlus is converting the unit for precipitable water in TMY2 files from mm to cm.  The NREL documentation for TMY2 states mm, although I've not been able to recover a TMY2 file and confirm.

tylunel, 2019-09-30:
My weather input is as follows : 
pvlib.iotools.epw.read_epw('https://energyplus.net/weather-download/north_and_central_america_wmo_region_4/USA/MA/USA_MA_Boston-Logan.Intl.AP.725090_TMY3/USA_MA_Boston-Logan.Intl.AP.725090_TMY3.epw')

The more general inputs of my model are in the attached text file (just replace .txt with .py to run with python).

I agree, in TMY2 files precipitable water is in mm. But in TMY3 files it is in cm according to the nrel user's manual.
The EnergyPlus doc states that data should be in mm (https://energyplus.net/sites/all/modules/custom/nrel_custom/pdfs/pdfs_v9.1.0/AuxiliaryPrograms.pdf , cf p.14). 
I guess EnergyPlus or NREL made an error about the unit when writing the doc and it resulted in bad conversion, because the epw file coming from tmy2 seems to be to ok (in cm), but not the epw files coming from tmy3 (in mm)

tylunel, 2019-09-30:
The issue can be solved just by adding two lines in pvlib.iotools.epw.read_epw():

    if 'tmy3' in filename.lower():
        data['precipitable_water'] = data['precipitable_water']/10

Nevertheless, if the issue comes from an error in the conversion of epw files by EnergyPlus, maybe it will be corrected soon, and these two lines will have to be removed.

roelloonen, 2019-09-30:
Yes, what @tylunel writes is in line with my understanding. 

The proposed modification seems to work. A drawback is that it makes an exception for TMY3, which is arguably among the most commonly used data sources for EPW, but certainly not the only one: https://energyplus.net/weather/sources

I will contact one of the developers of the EPW format. Maybe he can shed some more light on this issue.

cwhanse, 2019-09-30:
OK, I'm thoroughly confused.  Facts:

- EnergyPlus documentation says mm is the unit for precipitable water.
- TMY2 documentation says mm was used.
- TMY3 documentation says cm is currently used.
- pvlib functions expect cm.

@tylunel note indicates that TMY2 files obtained in EPW format have the units in cm, and TMY3 files in EPW format have the units in mm.  This seems backwards, but inspecting the files @tylunel linked supports this description. 

Does EnergyPlus have the units switched for TMY2 and TMY3 in EPW format?

cwhanse, 2019-09-30:
> The issue can be solved just by adding two lines in pvlib.iotools.epw.read_epw():
> if 'tmy3' in filename.lower():
>     data['precipitable_water'] = data['precipitable_water']/10

cwhanse, 2019-09-30:
sorry misclicked.

roelloonen, 2019-10-03:
> I will contact one of the developers of the EPW format. Maybe he can shed some more light on this issue.

The feedback I got is that the precipitable water in the EPW files based on TMY3 was indeed converted incorrectly.
There is apparently no possibility to change the data on www.energyplus.net, but the corrected TMY3 files on http://climate.onebuilding.org/ will be reposted. 

As a consequence, there will soon be corrected TMY3.epw files and original/"wrong" TMY3.epw files. In this case, the correction based on filename is no longer a failproof solution.

Would there maybe be a way to infer from the precipitation data which of the two units is clearly the most realistic one? Or perhaps that adds way too much complexity for the problem we are trying to solve?

cwhanse, 2019-10-03:
I don't think that the `read_epw` file should attempt to correct errors in EPW data files. I think that will only cause trouble for pvlib down the road.

I'm OK if we want to add a Note to the docstring about TMY3 files from energyplus.net having a known error, and providing some helpful suggestions for correcting it.

tylunel, 2019-10-03:
As far as I know, precipitable water data is only used as input in atmosphere.first_solar_spectral_correction() and in atmosphere.kasten96_lt(). atmosphere.first_solar_spectral_correction() contains a kind of pre-process on precipitable_water data at the beginning of the function :
    # *** Pwat *** (=precipitable water)
    # Replace Pwat Values below 0.1 cm with 0.1 cm to prevent model from
    # diverging"

    if np.min(pw) < 0.1:
        pw = np.maximum(pw, 0.1)
        warn('Exceptionally low Pwat values replaced with 0.1 cm to prevent' +
             ' model divergence')

    # Warn user about Pwat data that is exceptionally high
    if np.max(pw) > 8:
        warn('Exceptionally high Pwat values. Check input data:' +
             ' model may diverge in this range')

Maybe we could add a line to change the highest values as it is done for value lower than 0.1 ? It would give the simple following line :
 pw = np.minimum(pw, 8)

This way the model will not diverge and will not return impossible spectral mismatch factor (like negative). Still the result will be wrong, but I guess it is better than getting impossible factor resulting in impossible effective_irradiance at the end.
Then with the warning and the adding in the docstring, it may be enough for the user to understand that something is wrong.

cwhanse, 2019-10-03:
> Maybe we could add a line to change the highest values as it is done for value lower than 0.1 ? It would give the simple following line :
> pw = np.minimum(pw, 8)

I'd prefer to add a check and warning to `read_epw`, because we know there are EPW files with bad values that aren't likely to go away.

I'd rather not overwrite values inside the spectral mismatch function, the warning seems sufficient to me.

wholmgren, 2019-10-03:
I think it's reasonable to add a `pw = np.minimum(pw, 8)` line to `first_solar_spectral_correction`. I also think it's reasonable to return nan for values outside the limits of the algorithm. I'm not inclined to change the `read_epw` function, and I'm neutral on changing its docstring.

cwhanse, 2019-10-03:
> I think it's reasonable to add a pw = np.minimum(pw, 8) line to first_solar_spectral_correction.

OK if we provide `pw_min` and `pw_max` as kwargs, if so we should add kwargs for the air mass limits as well.  I am reluctant to second-guess what a user wants from the function.

> I also think it's reasonable to return nan for values outside the limits of the algorithm.

That's a challenge. The original paper developed and validated for 0.1 <=pw <= 5, but who's to say the model is invalid outside that range.

cwhanse, 2019-10-09:
@wholmgren and I talked about this in person (still happens sometimes :).  Neither of us are strongly opposed to any of the following options:

1) add a note to `read_epw` about older TMY files from energyplus.net
2) add `pw = np.minimum(pw, max_pw)` to `first_solar_spectral_correction` and expose `max_pw` as a kwarg. Could use this opportunity to also implement `min_pw`, `min_am` and `max_am` but that could be done in a later PR.  Default of 8. Return `np.nan` when inputs are outside the accepted `pw` range, and `warning`.

We both think the better course for `read_epw` is to return exactly what is in the EPW file, rather than attempt to check or correct the content.

PR #816 (2019-11-14): fix for *_TMY3.epw reading issue #782
This PR is for solving epw file reading problem.
A note is added in docstring of epw.py and a validity check is added in atmosphere.first_solar_spectral_correction. This validity check is done according to kwargs min_pw and max_pw. If the value is higher, np.nan is returned.

pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with your code at any stage of completion.

The following item [... cut]
closed 2019-11-14 by cwhanse