pvlib/pvlib-python #785: pvsystem.sapm error with input `module` as type `DataFrame`

cwhanse, 2019-10-03:
**Describe the bug**
Docstring states 

```
    module : dict-like
        A dict, Series, or DataFrame defining the SAPM performance
        parameters. See the notes section for more details.

```

DataFrame input is not tested. The function errors when module parameter are supplied in a DataFrame. 

**To Reproduce**
Steps to reproduce the behavior: see attached

**Expected behavior**
`pvsystem.sapm` should give correct results if parameters are in a DataFrame. Otherwise, we should remove the option from the docstring.

**Versions:**
 - ``pvlib.__version__``: 0.6.1
 - ``pandas.__version__``: 0.24.2
 - python: 3.6.8

[sapm_dataframe.txt](https://github.com/pvlib/pvlib-python/files/3687135/sapm_dataframe.txt)

wholmgren, 2019-10-03:
Is there a compelling reason to support a DataFrame input? If not, let's remove the option. I'm ok continuing to support Series because that's what you get when you select a column from the module parameters DataFrame.

cwhanse, 2019-10-03:
The only use case I can think of is to calculate output for several modules with the same irradiance and temperature condition. It would be convenient to generate a table of output at STC, for example. Important?  I'm not convinced.

wholmgren, 2019-10-03:
Sounds like a problem for a for loop or `DataFrame.apply(axis=1)`

closed 2019-11-07 by wholmgren