pvlib/pvlib-python #786: fix documentation references to inverter/module data

wholmgren, 2019-10-04:
The CEC data update in #767 broke the documentation. At least the [ModelChain](https://pvlib-python.readthedocs.io/en/latest/modelchain.html) page is broken, maybe more.

cwhanse, 2019-10-04:
Scanning the pages, these are also broken:
[PVSystem attributes](https://pvlib-python.readthedocs.io/en/latest/pvsystem.html#pvsystem-attributes)
[PV Power Forecast](https://pvlib-python.readthedocs.io/en/latest/forecasts.html#pv-power-forecast)

birgits, 2019-10-08:
In the course of fixing this it would also be nice to [fix the link](https://github.com/pvlib/pvlib-python/blob/master/pvlib/pvsystem.py#L1803) to where the original files are downloaded from. I would have fixed it but could not find where the data sets are hosted. 
Further, it would be great to provide a link to some documentation of the data sets in order for a user to figure out the meaning and units of the provided parameters. I think this link would be helpful:
https://sam.nrel.gov/photovoltaic/pv-sub-page-2.html

cwhanse, 2019-10-10:
> In the course of fixing this it would also be nice to fix the link to where the original files are downloaded from. I would have fixed it but could not find where the data sets are hosted.

Thanks for pointing out this old link. The current [link](https://github.com/NREL/SAM/tree/develop/deploy/libraries).

> Further, it would be great to provide a link to some documentation of the data sets in order for a user to figure out the meaning and units of the provided parameters. I think this link would be helpful:
> https://sam.nrel.gov/photovoltaic/pv-sub-page-2.html

Agree, and again, thanks!

PR #791 (2019-10-18): Fix documentation references to inverter/module data
 - [x] Closes  #786 
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is sufficient.  [... cut]
closed 2019-10-18 by wholmgren