pvlib/pvlib-python #789: rethink module-level model parameters dictionaries

wholmgren, 2019-10-16:
#783 is adding a `iam.IAM_MODEL_PARAMS` dict that's simply sets of required parameters for IAM models. 

#682 added `temperature.TEMPERATURE_MODEL_PARAMETERS` that includes default values.

#548 added `pvsystem.DC_MODEL_PARAMS` that's also only sets of required parameters.

We should reconsider if we want to expose these dictionaries to users.

These dictionaries are used by methods in `PVSystem` and `ModelChain` to automagically determine what functions to call. In principle, user code might do something similar with them, but I doubt that happens.

The `TEMPERATURE_MODEL_PARAMETERS` dictionary with default values is probably the most useful to an end user because the default parameters are useful when calling `temperature` module functions directly (i.e. not through ModelChain/PVSystem).

My short term recommendation is to:

* deprecate `pvsystem.DC_MODEL_PARAMS` into `pvsystem._DC_MODEL_PARAMS`. I'd also be ok with just changing it without deprecation because 1. 0.7 is a big release and 2. this change won't effect many people and 3. My time is limited.
* change `iam.IAM_MODEL_PARAMS` to `iam._IAM_MODEL_PARAMS` to prevent problems before they start
* leave `TEMPERATURE_MODEL_PARAMETERS` alone

veronicaguo, 2019-10-30:
Hi Will, I opened a short PR to make `DC_MODEL_PARAMS` and `IAM_MODEL_PARAMS` private. Please let me know what you think and if I missed anything. Thanks!

wholmgren, 2019-10-30:
Thanks @veronicaguo! I'm hoping that others will let us know what they think of these API choices and if the public dictionaries are actually useful to them or not...

PR #805 (2019-11-13): Don't expose model parameter dictionaries to users
 - [x] Closes  #789 
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
 - [x] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
 - [x] Code quality and style is sufficient. Pa [... cut]
closed 2019-11-13 by wholmgren