pvlib/pvlib-python #809: Pre-release announcement: pvlib-python v0.7

cwhanse, 2019-11-04:
We're close to release of pvlib-python v0.7. This will be a **major** release including (but not limited to):

- refactor temperature models to a new module `temperature.py` (API changes)
- refactor incidence angle modifier models to a new module `iam.py` (API changes)
- update module and inverter parameter files (new files have a different format)
- add a new module `ivtools.py` with functions for fitting diode models to data
- removal of some deprecated functions.

A full list of changes to date is [here](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/whatsnew/v0.7.0.rst).  

We plan to merge several open PRs for v0.7: #807, #805, #808, #804.  We may clean up some of the open issues that are currently marked for v0.7.

If you have an open pull request that still needs work, or are considering a pull request in the near future, please respond so that we can plan accordingly.

wholmgren, 2019-11-04:
Some of those PRs and associated issues are missing 0.7 milestone markers. 

Looking at [the 0.7 issues/PRs](https://github.com/pvlib/pvlib-python/milestone/23)... I'm ok punting all of these to a future release. #472 might be a good one to do sooner than later, though.

Looking at [the untagged issues](https://github.com/pvlib/pvlib-python/issues?q=is%3Aopen+is%3Aissue+no%3Amilestone)... Maybe add #785 and #768.

What's a realistic release date?

wholmgren, 2019-11-26:
I cleared a lot of stale issues/PRs from the 0.7.0 milestone. #807 is currently the only unfinished item marked for 0.7.0. I'm ok holding for that if it's going to be fast. I'm also ok with pushing 0.7.0 now and following up with 0.7.1 and maybe 0.7.2 over the next month or two before introducing breaking changes that mark the start of the 0.8 development cycle.

wholmgren, 2019-12-02:
@cwhanse this is the last issue labeled 0.7.0, but #821 is unlabeled and close to completion. Anything else?

cwhanse, 2019-12-03:
Let's move on releasing v0.7 once #821 is merged.

PR #839 (2019-12-18): update whatsnew.rst, add contributors, v0.7.0 release date, address #809
 - [x] Closes #809
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
* N/A
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
* N/A
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all chan [... cut]
closed 2019-12-18 by wholmgren