pvlib/pvlib-python #810: (Draft) Development plan for pvlib-python

cwhanse, 2019-11-04:
A small group of maintainers and developers drafted this [Development roadmap](https://github.com/pvlib/pvlib-python/wiki/pvlib-python-Development-Roadmap) for the next several releases of pvlib.  Contributions to all areas of pvlib are always welcome. 
We hope that the roadmap provides pvlib users with advance notice of several intended changes, and suggests priorities for pvlib developers.

Comments on the road map are welcome. Volunteers to take on development tasks are very welcome.

AdamRJensen, 2021-03-18:
@cwhanse Very nice with a roadmap, I'm particularly interested in the weather resources mentioned. First I can mention that I'm working on pull requests for PVGIS hourly radiation and SoDa-Pro CAMS and McClear. Also, a reader for BSRN files has been added.

Now, I think it would be very valuable to add support for retrieving MERRA-2 data. The Roadmap lists the provider as SoDa-Pro however, SoDa-Pro does not make MERRA2 available for free (unlike CAMS & McClear which are freely available through SoDa-Pro). I would imagine that it would be possible to get MERRA2 for free directly from NASA?

Also, what would be the use case for adding access to the World Bank Global Solar Atlas?

cwhanse, 2021-03-18:
@AdamRJensen I am not familiar with the MERRA2 dataset or its sources. @mikofski? I am confident there is interest in a reader for pvlib.

The World Bank solar data set hasn't been brought up before. It appears to be SolarGIS's data combined with a relatively simple PV output model, which also appears to be sourced from SolarGIS. One advantage is global coverage. I'm not opposed to a World Bank reader. But I suspect that the World Bank's data is of less interest to pvlib users than other irradiance data sources.

AdamRJensen, 2021-03-18:
@cwhanse I do believe that you are right in that the World Bank Global Solar Atlas is compiled from SolarGIS data, however, the atlas only provides annual average values and not any time-serie data. This is why I question its usability for pvlib users and perhaps that it should be removed from the roadmap.

cwhanse, 2021-03-18:
@mikofski your thoughts on the usefulness of the World Bank solar data? I think you had added it to the roadmap.

mikofski, 2021-03-18:
Sorry, tbh i'm not using world bank data. I think a coworker recommended it.

cwhanse, 2021-03-18:
I'll downvote it's relevance. Let's put off any effort to add a reader for the World Bank data set.

AdamRJensen, 2021-07-21:
@mikofski NSRDB is listed as incomplete in the roadmap, speficially "Individual site years (1991-2010) data" is missing.

@kanderso-nrel and I had a look and that data seems quite outdated and probably only available in one large zip file.

Are we still interesting in somehow adding this to pvlib or can it be removed from the roadmap?

mikofski, 2021-07-22:
Good points, I don't know what the best path is, or what others think. I think there's a lot of value in this data, and there's really no other alternative. I am strongly in favor of timeseries analysis over TMY, but you're right that it's starting to be old data (>10 years) and if it's in a zip file, no clear path to how we manage it, I don't think bundling is an option.

I agree, let's cross it off the list if everyone else agrees it's not worth it, or we can move to "won't do" and perhaps in the future we'll find somewhere the data is hosted and we can reconsider?

AdamRJensen, 2021-07-23:
@mikofski Out of curiosity, what makes you say that it has a lot of value? I would imagine that NREL's current NSRDB PSM3 has better coverage and is more accurate, hence they have stopped servicing it and put it in a zip archive.

mikofski, 2021-07-23:
PSM3 tmy has consistently shown a high bias, not everywhere, but on average, at least for the projects I've worked on. I don't know if this extends to the 5 min time series data, but it seems likely.  On the other hand tmy3 has been consistently closer to the median of other collocated solar resources

AdamRJensen, 2021-07-28:
@mikofski @kanderso-nrel I met up with Manajit from NREL (who's one of the people behind the NSRDB) yesterday and brought up this topic. He was quite clear in that NREL prefers pvlib not to provide data retrieval functions for the older datasets as PSM3 is significantly more accurate and the older versions are confusing users. Unless anyone objects I'll remove "Individual site years (1991-2010) data" from the roadmap. Do also note that PSM3 also supports the generation of TMYs.

cwhanse, 2021-07-28:
I don't agree with removing the capability to access the older data, some people still use it for historical purposes or because a previous analysis is conditioned on that data. If NREL doesn't want people to use that data, they could remove it from their servers.

AdamRJensen, 2021-07-28:
Perhaps my post was not clear. I certainly don't intend to remove any existing functionality. But I am suggesting not adding retrieval capabilities (a get function) for the "Individual site years (1991-2010) data" - hence removing the unticked box on the roadmap. The data is stored in one large zip file, so it isn't easily accessible anyways.

cwhanse, 2021-07-28:
Thanks for the clarification - I should have looked at the roadmap first. Are you saying that we will not add a get function to retrieve the 'MTS3' data indicated [here](https://nsrdb.nrel.gov/about/u-s-data.html)?

mikofski, 2021-07-30:
I'm fine with removing from the roadmap retrieval of the older timeseries data that was used to generate TMY2 and TMY3, but I disagree with the assertion that we should stop using this data, and I haven't seen any reliable proof that the PSM3 data sets are significantly better. However, I don't see logistically how we can store the old datasets.