pvlib/pvlib-python #813: Please add a download counter to the GitHub readme

jsstein, 2019-11-05:
**Is your feature request related to a problem? Please describe.**
Not a problem, but an opportunity

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add download shield to Readme.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

CameronTStark, 2019-11-05:
Josh asked me about this  and I found a source for PyPI and conda-forge.

Here are examples from each:
PyPI Downloads: [image]
conda-forge Downloads:  [image]

I'd also suggest tabularizing these shields to organize them [like pandas does](https://github.com/pandas-dev/pandas#pandas-powerful-python-data-analysis-toolkit). This would help the uninitiated to be clear about what each of these might mean. Certain people will be primarily concerned with code quality while others might be interested in publications or impact based on raw download counts.

I can make a PR for this as an example if there's no objections.

wholmgren, 2019-11-06:
Sounds great, thanks!

PR #820 (2019-11-15): Add shield organization table with download shields
 - [x] Closes  #813
 - [x] I am familiar with the [contributing guidelines](http://pvlib-python.readthedocs.io/en/latest/contributing.html).
 - [x] Fully tested. Added and/or modified tests to ensure correct behavior for all reasonable inputs. Tests (usually) must pass on the TravisCI and Appveyor testing services.
* N/A
 - [x] Updates entries to `docs/sphinx/source/api.rst` for API changes.
* N/A
 - [X] Adds description and name entries in the appropriate `docs/sphinx/source/whatsnew` file for all changes.
* N/A
 - [x] Code quality and style [... cut]
closed 2019-11-15 by wholmgren