pvlib/pvlib-python #825: Consistent names for reference kwargs

cwhanse, 2019-11-25:
**Is your feature request related to a problem? Please describe.**
Both `irrad_ref`and `reference_irradiance` are used for the reference irradiance kwarg for effective irradiance and DC models.

**Describe the solution you'd like**
Suggest it is worth changing to `reference_irradiance` throughout, and to replace `temp_ref` by `reference_temperature`.

adriesse, 2019-12-06:
Consistency would be great.  I'm not a fan of really long names, but you already knew that.

adriesse, 2019-12-06:
I just came across `T0` and `E0` in `pvsyst_parameter_estimation`...

cwhanse, 2020-09-04:
@wholmgren what is your view of the benefit of this proposed change?

wholmgren, 2020-09-04:
@cwhanse I'm +1 on changing them. I like your suggestions, but here are some quick searches for consistency... 

I see pvterms has a different set of suggestions, `irradiance_ref` and `temperature_ref` (I'm not committing to using pvterms but I am committing to checking it). I see `reference_irradiance` is a keyword argument in `PVSystem.sapm_effective_irradiance` though it doesn't do anything. I don't see `*reference*` elsewhere in the library. A handful of parameters and functions use `*_ref`. 

To @adriesse's point, I'm a fan of really long names, but you already knew that.

wholmgren, 2020-09-04:
@cwhanse there's also the `ivtools.utility.constants` dict

```
constants = {'E0': 1000.0, 'T0': 25.0, 'k': 1.38066e-23, 'q': 1.60218e-19}
```

echedey-ls, 2024-12-22:
It seems like we already have the tools to fix this issue (the variable naming table, and the deprecated kwarg decorator) and it would be a great benefit.

I think this issue deserves a vote on where the `reference` indicator should be in a parameter name (independently of the last trends):
- 👍  : `*_reference`
- 🎉 : `reference_*`
- 🚀 : `*_ref`
- 👀 : `ref_*`

cwhanse, 2024-12-22:
Either 👍  or 🚀  are OK with me. I prefer placing the adjective 'reference' after the noun 'irradiance'

adriesse, 2024-12-22:
Let's allow for exceptions to the rule, if something sounds awkward.