pvlib/pvlib-python #828: adopt NEP-29

mikofski, 2019-12-04:
**Is your feature request related to a problem? Please describe.**
the travis-ci test fails for #823 because the minimum NumPy version for py35-min is version 1.10.4 - the `np.roll()` function used was updated to allow multiple axes in NumPy-1.12

**Describe the solution you'd like**
[NEP-29](https://numpy.org/neps/nep-0029-deprecation_policy.html) is a proposal by maintainers of IPython, Matplotlib, SciPy, and NumPy that proposes the support window for Python and Numpy.

TL;DR: in the immediate future, drop support for Python-3.5 and Numpy<1.14

**Describe alternatives you've considered**
- statically create the 3 rotation matrices, removing the use of `np.roll` completely
- use the pre-NumPy-1.12 version of `np.roll`

**Additional context**
Since many other major packages that are used by pvlib are conforming to NEP-29, it makes a lot of sense for pvlib to also conform.

wholmgren, 2019-12-04:
I think NEP-29 is too aggressive. I'm fine with adopting NEP-29 rules for *expected* compatibility. But NEP-29 seems to go a step farther and suggest that every downstream project release should only support the prescribed versions regardless of if it's actually necessary. In the specific case above, why should we restrict pvlib to numpy 1.14 when numpy 1.12 and 1.13 will work just as well? 

I suggest that we instead aggressively drop numpy/python versions when 1. there is a reason to do so and 2. the versions are outside the NEP-29 windows.

mikofski, 2019-12-04:
an alternate approach would also be to create a `requires_numpy_1_12` decorator in `conftest.py`

wholmgren, 2019-12-04:
better to just bump the requirement.

PR #830 (2019-12-13): update to numpy-1.12.0

 - [x] Closes #828
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` ` [... cut]
closed 2019-12-13 by wholmgren