pvlib/pvlib-python #837: add call outs to all references in docstrings

wholmgren, 2019-12-13:
#833 converts all references into the numpydoc format. The unfortunate side effect of this is that it creates a lot of warnings in the sphinx build because many of the references are not called out in the docstring. #833 silences these warnings using `suppress_warnings = ['ref.footnote']`. In the long run, it would be great to call out the references and remove the suppress warnings statement. That would move us one step closer to automatically and comprehensively checking docstrings for style.

souravsingh, 2020-01-07:
@wholmgren I am interested in taking a stab at the issue.

wholmgren, 2020-01-07:
Great, thanks, let us know if you need anything to get started!

cwhanse, 2020-01-08:
@wholmgren @souravsingh related to this, one issue with references, is that we are formatting references as footnotes rather than citations,, since we are using a numeral as the first character.  Compare the [footnote ](https://external-builds.readthedocs.io/html/pvlib-python/708/generated/pvlib.ivtools.fit_sde.fit_cocontent.html#pvlib.ivtools.fit_sde.fit_cocontent) format to the [citation](https://external-builds.readthedocs.io/html/pvlib-python/708/generated/pvlib.ivtools.fit_sdm.fit_desoto.html) format.

If we switch to citations, I believe it is [possible ](https://www.sphinx-doc.org/en/1.5/rest.html#citations) to collect references in a common source file. Worth discussing if we want to do that; it would save a lot of repeated docstring text.

wholmgren, 2020-01-09:
@cwhanse is the only change that we use strings (e.g. `_[myref]`) instead of integers (e.g. `_[1]`)? I don't see any other difference. 

>  it would save a lot of repeated docstring text.

I can see some pros (less repetition, all references in one place) and cons (references are farther from the relevant code), but I think I'd go for it. Do you know of other projects that take this approach?

cwhanse, 2020-01-09:
scipy uses both styles (footnote and citation), e.g. [nonlin.py](https://github.com/scipy/scipy/blob/master/scipy/optimize/nonlin.py) it appears that the style is up to the implementer.  scipy puts citations with each function, and repeats the docstring text where the citation applies to more than one function. The scipy docs style avoids formatting numbered citation callouts as superscripts (footnote style), e.g. [newton-krylov](https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.newton_krylov.html#scipy.optimize.newton_krylov) 

I don't have a strong preference for either [1] or [Perez90]. It would be nice to not see callouts as superscripts, but that's a nice-to-have.
 
I'm now -1 on collecting references into a common file. That may be advantageous for documentation pages, but I see removing the references from each function is a step backwards and isn't worth the small gain of removing repeated text from docstrings.

wholmgren, 2026-10-29:
closing so there is one fewer issue to attract AI spam

closed 2026-10-29 by wholmgren