pvlib/pvlib-python #838: Parameter descriptions for retrieve_sam

toddkarin, 2019-12-13:
The pvlib.pvsystem.retrieve_sam function pulls information on modules, but the different keys don't have descriptions. For example, I don't know what 'A_c' means in the CECMod database. Can these various fields be described?

toddkarin, 2019-12-13:
By sleuthing into the CEC database 'A_c' seems to be the module area, but it would still be nice to have this documented in PVLIB.

toddkarin, 2020-07-24:
@cwhanse Would this be a good opportunity for incorporating [pv-terms](https://duramat.github.io/pv-terms/) in order to describe the various parameters? It may also require a relabeling of the CEC values.

cwhanse, 2020-07-27:
Yes, it is timely to consider changing terms and improving the documentation. There's a [discussion](https://github.com/pvlib/pvlib-python/issues/964) of refactoring the current `retrieve_sam` into the `pvlib.iotools` module. With new functions we'll deprecate the current `retrieve_sam` for a while, to make transition easier.