pvlib/pvlib-python #841: Add ability to read local NSRDB PSM3 CSVs

kanderso-nrel, 2019-12-19:
Unless I've missed something in the docs, the only pvlib function that parses PSM3 CSVs is `iotools.get_psm3`, which requires making a network request to the NSRDB PSM3 API.  It would be nice to be able to read local PSM3 files (e.g. those downloaded manually from the [NSRDB Viewer](https://maps.nrel.gov/nsrdb-viewer/)) in addition to requesting them from the web API.  Here are a couple proposals:

- Break out the parsing code in `iotools.get_psm3` into its own function `iotools.read_psm3`, leaving `get_psm3` as a networking wrapper around the PSM3 API that then calls `read_psm3` on the API response.  
- Modify `get_psm3` to check if the input is a filename instead of API parameters and retrieve data as appropriate. 

I'd favor the former since it separates networking code from parsing code.  Some other `iotools.read_*` functions can make web requests, but those are just simple urls and more or less equivalent to reading a local file whereas `get_psm3` has to construct a full API request.

cwhanse, 2019-12-19:
Does a PSM3 file for a specific year follow the TMY3 format? I think the PSM3 also delivers TMY3 files, if requested. If it doesn't, then I'd say yes, a `read_psm3` function would be of value.

kanderso-nrel, 2019-12-19:
I believe the both the NSRDB Viewer and its API export single-year and TMY data in the SAM CSV format, which is what `get_psm3` is currently parsing.  I took a look and didn't see an option to change the output format in the web interface, but if there a way to export in TMY3 format then I'm happy to just use that. 

It occurs to me that a better name might be `read_sam_csv` or similar instead of `read_psm3`.

cwhanse, 2019-12-19:
My earlier comment is an unclear mess. Let me try again.

Does a PSM3 file for a specific year follow the TMY3 format? If the format is not TMY3, then I'd say a separate function to read that format could be of value.  I'm ambivalent about the name, although `read_sam` could be interpreted as not intended to read NSRDB files. 

You can get TMY3 files from the NSRDB by passing `names='tmy'` although this is not mentioned in the [NSRDB documentation](https://developer.nrel.gov/docs/solar/nsrdb/psm3_data_download/#data-file-format).

kanderso-nrel, 2019-12-19:
> You can get TMY3 files from the NSRDB by passing names='tmy'

I think we need to distinguish TMY3 data from the TMY3 format.  It is true that TMY data can be retrieved from the PSM3 API, but I believe it is returned in the SAM CSV format, not the TMY3 format.  

> Does a PSM3 file for a specific year follow the TMY3 format?

No, and furthermore (as far as I am aware), all PSM3 data (both single-year and TMY; both web viewer and API) is returned in the SAM CSV format.  

[1] https://nsrdb.nrel.gov/about/tmy.html (scroll to "Format of TMY Data")
[2] https://nsrdb.nrel.gov/about/u-s-data.html (ctrl-F "Data Header")
[3] https://rredc.nrel.gov/solar/old_data/nsrdb/2005-2012/wfcsv.pdf

cwhanse, 2019-12-20:
> all PSM3 data (both single-year and TMY; both web viewer and API) is returned in the SAM CSV format.

Thanks; this clarifies that a new `read_xxx` function for this format would not duplicate `pvlib.tmy.read_tmy3`.

PR #842 (2020-01-03): refactor get_psm3 code into parse_psm3, read_psm3
 - [x] Closes #841 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub I [... cut]
closed 2020-01-03 by wholmgren