pvlib/pvlib-python #847: Equation of time Spencer coefficient

IntiPic, 2020-01-06:
There's a difference between Iqbal's equation of time and the code. The 0.0000075 coefficient in the code has one more zero than Iqbal's. It should be changed into 0.000075.
I'm new at github so I don't know how is the process for reporting these issues.

Thanks, bye!

cwhanse, 2020-01-06:
@IntiPic this is where issues are raised, thank you. In this case, the coefficient 0.0000075 (five '0's) is a correction as described in the [documentation ](https://pvlib-python.readthedocs.io/en/stable/generated/pvlib.solarposition.equation_of_time_spencer71.html#pvlib.solarposition.equation_of_time_spencer71) for that function.

cwhanse, 2020-01-06:
Leaving this open since I think the docstring for `pvliv.solarposition.equation_of_time_spencer71` can be improved, in particular, the link to the email archive describing the correction is not referenced and so doesn't appear on the documentation page.

IntiPic, 2020-01-06:
Thanks a lot. I should've read it better.
This is a big help. Been using the four 0's version for quite a while without knowing this.

mikofski, 2020-01-06:
You may click the link in the documentation to be redirected to the [Fourier paper](https://www.mail-archive.com/sundial@uni-koeln.de/msg01050.html) post by Mac Oglesby in the Sundial mailing list

[image]

I wonder if we should just copy this letter into either PVPMC archive or this git repo for posterity in case it is lost or removed?

wholmgren, 2020-02-13:
@cwhanse do you still want to improve the documentation of this function or can we close this?

cwhanse, 2020-02-13:
Ok to close, the docstring is sufficient.

closed 2020-02-13 by cwhanse