pvlib/pvlib-python #848: ENH: MAINT: STY: group iotool tests into folder

mikofski, 2020-01-07:
**Is your feature request related to a problem? Please describe.**
there are so many tests now, would it be okay to group some of them, eg iotools into a folder?

**Describe the solution you'd like**
move all of the iotools tests into a folder called `iotools_tests` inside the `pvlib/test` folder

**Describe alternatives you've considered**
just keep letting the number of files in test grow until someone else thinks it might be useful to group them

**Additional context**
[pandas/tests](https://github.com/pandas-dev/pandas/tree/master/pandas/tests) are grouped in folders, but forgo the `_tests` suffix, it does seem redundant, but will pytest still work without it? I'm not set on any particular convention, as long as it's consistent

so what do folks think? OK to post here? Should I have posted in pvlib-python google group instead? thx!

wholmgren, 2020-01-07:
I’m ok with any of these reorganizations of the tests. I have a mild preference for a pattern like pvlib/iotools/tests. 

And this is the right place to discuss.

mikofski, 2020-01-07:
I'm not completely sure, but I think all of the tests need/should to be in the same top level folder, or at least in the past, I've had problems with VSCode, so maybe it's not a pytest problem per sé.

So like `pvlib/test/iotools/test_tmy3.py` etc. would that be okay?

wholmgren, 2020-01-07:
Ok but I can assure you that pytest and vscode work just fine using the
pattern I suggested. There is something to be said for putting all the test
code in one place and something to be said for keeping it closer to the
code it tests.

cwhanse, 2020-01-07:
> I’m ok with any of these reorganizations of the tests. I have a mild preference for a pattern like pvlib/iotools/tests.

I would prefer pvlib/iotools/test over pvlib/test/iotools/, but not a strong preference.

mikofski, 2020-01-14:
The internet has spoken?

[image]

PR #859 (2020-01-16): reorganize tests into subfolders and use pathlib for conftest DATA_DIR
* closes #848
* create pvlib/tests/iotools and move all iotools tests into that
subfolder
* use fixtures in test_ecmwf_macc.py for expected_test_data
* use conftest.data_dir instead of redundant boilerplate process of
retrieving test folder for each tests, mostly for iotools
* fix test_ivtools.py was using full path to conftest but tests is not a
python package, so no
* change "pvlib/test" -> "pvlib/tests" to conform to popular convention

 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthed [... cut]
commit bb081ed2 (2020-01-16): reorganize tests into subfolders and use pathlib for conftest DATA_DIR (#859)

* reorganize tests into subfolders

* closes #848
* create pvlib/tests/iotools and move all iotools tests into that
subfolder
* use fixtures in test_ecmwf_macc.py for expected_test_data
* use conftest.data_dir instead of redundant boilerplate process of
retrieving test folder for each tests, mostly for iotools
* fix test_ivtools.py was using full path to conftest but tests is not a
python package, so no
* change "pvlib/test" -> "pvlib/tests" to conform to popular convention

* in azure pipelines, pytest pvlib only,  [... cut]
closed 2020-01-16 by wholmgren