pvlib/pvlib-python #849: getter/parser for PVGIS hourly-radiation

mikofski, 2020-01-08:
**Is your feature request related to a problem? Please describe.**
Related to #845 

**Describe the solution you'd like**
Similar to `get_pvgis_tmy` retrieve pvgis hourly radiation data from their api

**Describe alternatives you've considered**
Pvgis is becoming a popular resource more and more people are asking me for it, it is nice because it is a global collection of several different radiation databases including nsrdb and others, and different from cams, the data is complete, ie it has sure temperature, wind speed as well as all 3 components of irradiance

**Additional context**
This would be part of the `iotool` sub-package. There's already a `pvgis.py` module with a getter for tmy be data

AdamRJensen, 2021-02-28:
@mikofski I'm developing a function for PVGIS hourly radiation and have a question to the person who developed the TMY function.

The get_pvgis_tmy function uses io.BytesIO(res.content) for the csv outputformat, whereas the epw format decodes the response: io.StringIO(res.content.decode('utf-8')). Is there a reason for not decoding the response for the csv outputformat (performance)? The reason I'm asking is that my code for _parse_pvgis_hourly_csv is much cleaner when the file-buffer has been decoded first.

mikofski, 2021-02-28:
Sorry, no good reason I can think of.  I was probably just hacking at it until I could get it to work.  Maybe I tried a string buffer first and had issues? I can't find your code for `_parse_pvgis_hourly_csv`, are you planning to open a pull request to close this issue? Would be great to clean up my mess with cleaner code if you have it. Thanks!

AdamRJensen, 2021-02-28:
@mikofski Thanks for the update, it could have been some smart trick I wasn't aware of. I'm opening a pull request within the next few days to add functions for PVGIS hourly radiation and then in a second pull I can clean up the TMY so they follow similar formats.

PR #1186 (2021-07-14): Add getter/parser for PVGIS hourly-radiation

 - [x] Closes #849
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue  [... cut]
closed 2021-07-14 by kanderso-nrel