pvlib/pvlib-python #857: SMARTS wrapper function

shirubana, 2020-01-15:
Back in 2018 I sent @cwhanse a Matlab wrapper for SMARTS we used on our UofA laboratory, with permission to publish, but got sidetracked and never documented. I just ported the code to Python and it seems to be working good. 

I'll be ding a pull request for the python wrapper.

This is the ` original matlab code_<(https://www.dropbox.com/s/qqmht1vdkq79l8p/SMARTS_MatlabWrapper.rar?dl=0)>

This where some of the highlights from the emails in 2018:
- Basically [PVlib] can publish the Matlab code and put the responsibility to obtain SMARTS on the user. 
- We need to have a separate license for the UofAz Matlab code. (BSD-3 License included in the dropbox folder)
- Users will be responsible to obtain a copy of SMARTS Fortran from NREL, honor it’s license, and build their own MEX file (they have to compile the Fortran code using a Matlab compatible compiler, a list and instructions are at https://www.mathworks.com/support/compilers.html).

Couple notes:
-Not sure how the license extends to the Python code
-Right now the functions assume that the SMARTS executable ("smarts295bat.exe") is on the same folder as the function (pvlib folder). This should change depending on how the above wants to be handled, I'd appreciate support for that based on what we decide.

This is kinda related to #71 , but PyRTM had other functionalities as well.

cwhanse, 2020-01-15:
Thanks @shirubana for this initiative.

Would the python code you are preparing need a license separate from pvlib-python? If yes, can you post the license conditions? I suppose the license could be communicated by putting the conditions as comments in the python file, and adding a statement to the pvlib-python license. I'm not sure what conventional practice would be.

Perhaps the wrapper function can require the path to the SMARTS executable as an input.

cwhanse, 2020-01-15:
The license in the Dropbox (for the MATLAB wrapper, not for SMARTS) is the same as for pvlib-python, except for the copyright statement. I think we could put the copyright statement in your python code, and acknowledge the different copyright in pvlib-python's license file.

The SMARTS wrapper is pretty long (line count) with all the comments (which I would retain). I'd put it in it's own file smarts.py rather than adding to irradiance.py

shirubana, 2020-01-15:
Thanks for looking into this :) That sounds good to me. I'll make the changes.

mikofski, 2020-01-15:
super excited about this, thanks Silvana! I wonder if someone will take up [spectrl2](https://www.nrel.gov/grid/solar-resource/spectral.html) now?

shirubana, 2020-01-15:
Hmm... I haven't used that, should look into it. I have a process to go from the SMARTS spectra to spectrally resolved radiance for the cumulative sky for raytrace. That's the motivation behind making this more streamlined.

mikofski, 2020-01-15:
Ah, IMO [spectrl2](https://www.nrel.gov/grid/solar-resource/spectral.html) is much easier and faster to use than SMARTS, although it may not be as accurate? It's especially useful for inline spectral calculations, because it's limited only to 120 bands that matter, so you can accurately calculate spectral mismatch in real time instead of using a correlation or polynomial fit. There's already a python wrapper by SunPower called [SolarUtils](https://sunpower.github.io/SolarUtils/) but we've talked frequently about porting it to pure python to accelerate it using numba.

kanderso-nrel, 2020-06-11:
@mikofski or anyone else:  Just curious, do you know if there are any restrictions on the spectrl2 implementations listed at that [link](https://www.nrel.gov/grid/solar-resource/spectral.html)?  I don't see a license anywhere on that webpage or in the cited technical report, which also includes a code implementation.  Mostly I'm interested in whether the method needs to be implemented de novo or if we could just port one of those implementations.

PS I am aware of the irony of me asking you about NREL's licensing, no need to point it out...

shirubana, 2020-06-12:
i have a python version of spctrl2 coded, vut haven't tracked down the proper licensing steps for that one yet. it's in my to do...

Sent via the Samsung Galaxy S10, an AT&T 5G Evolution capable smartphone

mikofski, 2020-06-12:
I have been distributing a Python extension of [SOLPOS and SPECTRL2](https://pypi.org/project/SolarUtils/) for years. My view is if it's on the internet then it's public, but they specifically mention the [license & disclaimer](https://www.nrel.gov/disclaimer.html), so I distribute it with the package [here](https://github.com/SunPower/SolarUtils/blob/master/solar_utils/NREL_DISCLAIMERS-COPYRIGHTS-LICENSES.txt):

#### [SPECTRL2](https://www.nrel.gov/grid/solar-resource/spectral.html)
[image]

#### [SOLPOS](https://www.nrel.gov/grid/solar-resource/solpos.html)
[image]

In a nutshell the license and disclaimers say:
1. anyone can use the software freely
2. you may be subject to US laws
3. usage doesn't imply endorsement
4. there is no warranty or liability

IMO it's similar to a bsd-2 license

Silvana, AFAIK, this liberal license would allow your port to be a completely license free work. There's no copyright or attribution, in fact the license literally says you can use it anyway you want for free.
>paid-up [_sic_: free] nonexclusive, irrevocable world-wide license to reproduce, prepare derivative works, distribute copies to the public, and perform publicly and display publicly