pvlib/pvlib-python #861: Add a plotting wrapper that make it easy to get high-quality plots

UGuntupalli, 2020-01-20:
**Is your feature request related to a problem? Please describe.**
pvlib is great and thanks to all the great contributions that exist. I don't necessarily see this as a problem, but rather an enhancement to have a plotting wrapper library based on plotly/ bokeh that makes it easy to prepare high quality plots instead of using matlplotlib. I am happy to take a first pass. 

**Describe the solution you'd like**
Ideal solution would include good wrappers for generally used plotting functions like scatter plot, bar chart, line chart etc. with options to customize to allow the user to customize as needed, but also cut down up-front work in preparing the figure. 

**Describe alternatives you've considered**
This is just an enhancement, so I have not considered any alternatives, open to suggestions/ feedback.

kanderso-nrel, 2020-01-22:
Would these be new ModelChain methods that make plots based on the modelchain's attributes?  Or more general functions where you pass in the datasets of interest?  I think it might be helpful to show some pseudocode usage examples of what you're imagining, since right now your proposal doesn't mention any specifics about how this code relates to pvlib :)

At a high level I'd guess that the set of plots that pvlib users generate is going to be pretty varied, so any pvlib plotting functions would have to be very general and flexible to get much use.  If that's the case then I'm not sure what value they'd bring over the functions in the base plotting libraries themselves.  @UGuntupalli what do you think?

mikofski, 2020-01-22:
How about putting an example of making a plotly or bokeh plot in the new [examples gallery](https://github.com/pvlib/pvlib-python/tree/master/docs/examples)?

Here are my concerns about your proposal, IIUC:
* I believe that bokeh and plotly dependencies will be undesirable for most users
* IMO mpl and seaborn actually do make really excellent high-quality, publishable plots, and pandas plotting ability makes generating these already mostly builtin.  I disagree that bokeh and plotly are higher quality at least for publications. However I do agree that bokeh and plotly, and possibly altair, vega, and vega-lite might be better for websites, because those plotting libraries are more interactive and come with JS hooks -- but I think examples in the gallery will better serve users interested in making plots with these non-mpl/seaborn libs
* if the plotting wrappers are untestable, I'm concerned it will lower our coverage.

I'm sorry this is sounding negative. I really like this idea, and I do really encourage you to contribute, but IMHO I think this idea would be better in the examples gallery. Thanks

wholmgren, 2020-02-13:
Thanks @UGuntupalli for the idea and your efforts in #871. It looks like this is going to be too much of an infrastructure challenge for including in pvlib, but I do encourage you to pursue the examples elsewhere and advertise them to the pvlib community.

closed 2020-02-13 by wholmgren