pvlib/pvlib-python #864: v0.7.2 release planning

CameronTStark, 2020-01-22:
I'm opening this issue to centralize the discussion of what will be in the v0.7.2 release. Please try to centralize your thoughts and discussion about the next release here or by numerical reference from another issue/PR E.g. #XYZ. 

This is a new strategy proposed by @wholmgren and @cwhanse for someone assigned this role as "release manager" to track how current issues, PRs and bugfixes will add up to a release and ultimately perform that release. They've nominated me for this role for an undetermined amount of time so I'll do my utmost to stake this path through a continued dialoge with the community. 

I plan to continue to utilize the [v0.7.2 milestone](https://github.com/pvlib/pvlib-python/milestone/29) as a means of communicating what is planned to go into the next release. This issue will be a forum for the the community to voice support for certain enhancements to be prioritized for the release that will then inform the milestones. Of course this could be disrupted by important bug fixes that need a quick release like for v0.7.1 but bumping the milestone version to the next release shouldn't add to the disruption.  Overall I'll be reviewing the status of current issues and PRs for how they fit into a future release. 
 
_To start the conversation, please let me know what you think about what's currently listed on the v0.7.2 milestone and let me know if there's a reason to add more or remove something. Also, please suggest methods for making this process easier, faster or clearer._

Thanks for your support!

kanderso-nrel, 2020-02-12:
#764 got a lot of attention and seemed like it got close to the finish line but has been stagnant for several months.  It would be good to see it get cleaned up and finished.  I'd be happy to lead that if needed. 

An unrelated thought not specific to 0.7.2 but releases in general: creating the next whatsnew file in master whenever a release is finalized would avoid some awkwardness in the first couple PRs afterwards.  Maybe even create several whatsnew files ahead of time in preparation and just omit them from the main file until the time comes?

cwhanse, 2020-02-12:
I've been working on cleaning that up, I'll finish this week. I tried to contact @JPalakapillyKWH but got no response.

CameronTStark, 2020-02-12:
I just added #764 to the 0.7.2 milestone based on your statement @cwhanse. 

I am making myself an "Assignee" to issues/PRs as a means of organizing my personal "to-do" list but also a means of communicating to others that I've put myself on the hook to getting this done. I would like to encourage others to do the same so we know who to check in with on the current status or to restart activity on issues that have gone dormant.

mikofski, 2020-02-12:
I think we could wrap up #823, but it is pending reviews by Cliff and Kevin. There's also the question whether to use the term "system plane" or "slope[d] plane" for the sloped plane that contains all of the trackers axes.

But I see there are 14 open issues for v0.7.2, so maybe this one can wait until v0.8.

I'll try to take a look at #516 and #517 is no one else is already working n them?

cwhanse, 2020-02-12:
v0.8 for #823. I'm going to move #516 and #517 to v0.8 also, since they will change the API. #836 is a candidate for v0.7.2, it adds to (but doesn't otherwise change) the API.

mikofski, 2020-02-12:
I can probably knock out #880 then

CameronTStark, 2020-02-24:
I'm looking to shore up issues slated for 0.7.2 and to account for open issues tagged with 0.7.2. I'd like to open a discussion about where things need to go to get 0.7.2 released by Friday March 6th. 

I've worked to tag issues with the 0.7.2 milestone and you can view them all [here](https://github.com/pvlib/pvlib-python/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc+milestone%3A0.7.2). Below are my current assessments of issues tagged as 0.7.2.

The most pressing issues due to problems that affect usability/bugs are below. There hasn't been follow up on these so I'd like to highlight them here:
* [ ] #902 
* [ ] #889 

These PRs are on their way to being completed by then without concern.
* [ ] #880 - PR from @mikofski  
* [ ] #836 - PR from @kanderso-nrel 
* [ ] #771 - PR from @CameronTStark 

This issue appears to be closed. Can someone (maybe @cwhanse) confirm?
* [ ] #507 - Could be closed? 

The following issues have discussions that have stalled and if they aren't started now it's unlikely to be resolved in time for the release. If there are no activity in them I plan to push them back to 0.7.3. If they require more sweeping changes they could be set for 0.8.0.
* [ ] #891 @wholmgren 
* [ ] #765 @wholmgren 
* [ ] #716  @cwhanse 

[There are also many issues without milestones.](https://github.com/pvlib/pvlib-python/issues?page=1&q=is%3Aopen+is%3Aissue+no%3Amilestone+sort%3Aupdated-desc) If there are any issues that you believe should be part of 0.7.2 please highlight them here. 

I'm planning on tagging future issues/PRs with 0.7.3 unless there's an urgent bug fix or an easily completed enhancement PR.

mikofski, 2020-02-24:
I've pushed a PR for #889

cwhanse, 2020-02-24:
#507 is not closed. The discussion wandered to other issues e.g. #506 which was closed by #596 .

mikofski, 2020-02-28:
FYI: @CameronTStark 
* #907 is approved and I think ready to merge
* #910 is waiting for approval from @wholmgren but I believe I've addressed his most recent comments, and especially if there will be a v0.7.3 release then I think it would be better to put this temporary fix in place until v0.8 - I volunteer to maintain and refactor it for v0.8

CameronTStark, 2020-03-01:
FYI: I just updated [my "accounting" comment above](https://github.com/pvlib/pvlib-python/issues/864#issuecomment-590164373) and am interested in any feedback. 

I'm aiming to usher all 0.7.2 items through to completion ahead of creating a 0.7.2 release candidate version ahead of the final release.

cwhanse, 2020-03-02:
That leaves #764 still to finish, it's close, I'll get it done this week.

mikofski, 2020-03-25:
Hi @CameronTStark - can you give me any idea when the v0.7.2 will be released? The NumFOCUS March newsletter is going to press, and wondering if I should signal that we will have a new release or wait until April? thx

CameronTStark, 2020-03-25:
Hey @mikofski! Hope you're doing well. In short, I'm not inclined to rush the release at this time.

I've watched #934 making good progress and is probably the last change I was hoping for before wrapping up 0.7.2. With the transitions we've all been making in this time I'm sure many of us have been stretched thin, I know I have, so I don't want to assume its timeline. If @anomam and the team has availability to get that PR across the finish-line in time for the newsletter I'll do what I can to support them.

If there is disagreement I'm open to recommendations based on past precedent concerning waiting for #934 or moving forward.

cwhanse, 2020-03-25:
I think we have plenty of content now, and in tagged PRs, without #934.

anomam, 2020-03-30:
> Hey @mikofski! Hope you're doing well. In short, I'm not inclined to rush the release at this time.
>
> I've watched #934 making good progress and is probably the last change I was hoping for before [...]

Hey @CameronTStark and team, sorry I haven't made much progress on #934 lately. I can commit to integrating the requested changes this week if that helps. When is the planned date of release for v0.7.2?

CameronTStark, 2020-03-30:
Thanks @anomam for your continued help! This has been strenuous times so it's even more appreciated. 

No worries if you hit a delay but this should be good timing as I'm planning on merging the rest of milestone 0.7.2 and putting a release candidate out this week and the full release no more than a week later. This should give you a runway to hit either way!

CameronTStark, 2020-04-05:
I'm planning to wrap up v0.7.2 with the following steps:

* I'm ready to merge #937 and #938 but I don't want to steal PR cred and deal with the conflict myself.  Please let me know @cwhanse if you'd like me to handle the conflict and merge anyway.
* #948 is close to being done and will be merged.
* #947 is ready to merge by another maintainer when they have availability.
* After the above is completed I want to make a release candidate. If #934 and #936 are ready soon they can be integrated into the either the rc or the final release. 

I may add issues/PRs addressing the CI/CD functionality for increasing "quality of life" enhancements there.

Thank you all for your patience and continuing contributions!

cwhanse, 2020-04-06:
Think I've taken care of the conflicts for #937 and #938. Maybe one of us can complete #936? I could get to it Wednesday

cwhanse, 2020-04-16:
@CameronTStark anything I can do to help move forward with the release?

CameronTStark, 2020-04-16:
This isn't holding up the release but if you could review #934 we could check that off the list. 

Also, I've been wondering how the team has handled cases like #936 where the PR is done but the addition to What's New is hanging. Do maintainers have permissions to change the PR branch? If so how? I'll be glad to close it out if I know what to do.

Time allowing, I'll have the v0.7.2rc release out tomorrow regardless of what happens with these two. 

Thanks for checking in, I appreciate it.

wholmgren, 2020-04-16:
#939 is tagged for 0.7.2. Looks like a relatively easy and useful win if someone wants to take it on. Ok with me to proceed with the release though.

mikofski, 2020-04-16:
It's time for the May NumFOCUS newsletter. Can we release pvlib-0.7.2 in time to make the cut? My vote is to make #936 the last PR and release ASAP. In fact let's consider a fixed release cycle instead of a feature quota. EG: let's release every 6 months with whatever's in master at that time.

Cameron, It's looks like #953 was added to address the what's new for #936 
 
>Do maintainers have permissions to change the PR branch? If so how? I'll be glad to close it out if I know what to do.

The PR has to have the allow maintainers to edit PR box checked. It's on the right side of the PR at the bottom of the side bar.
[image]

But as a maintainer, you could always go rogue and pull rank. I'm fine with that as long as Will and Cliff are. IE: just merge the PR with a comment that says, "thanks and I'll update the what's new directly in master as an addon commit after." Then merge the PR, and make the changes directly in master and commit them. You can even edit, preview, & commit/push what's new directly in GitHub. No additional PR necessary IMHO. I've seen this paradigm a few times in other repos and I think it makes a lot of sense, especially for new contributors. IE: we don't always have to stick so rigidly to the feature-branch workflow.

CameronTStark, 2020-04-17:
> It's time for the May NumFOCUS newsletter. Can we release pvlib-0.7.2 in time to make the cut? My vote is to make #936 the last PR and release ASAP.

Yes! I was aiming for it so let's do it. 

> In fact let's consider a fixed release cycle instead of a feature quota. EG: let's release every 6 months with whatever's in master at that time.

I was thinking along the same lines given my recent experience. It of course doesn't need to be strict but it should set guidelines for when a release will be prepared.

> Cameron, It's looks like #953 was added to address the what's new for #936

I hadn't noticed this, thanks for highlighting. I'll merge both #936 and #953. I'll give amending the PR a try directly next chance arrives. Thanks again for guidance.

CameronTStark, 2020-04-20:
I created v0.7.2-rc.1 last Friday and am ready to push the full release out. #934 is on its way but there's pressure to release before the next NumFocus newsletter so I will perform the final release tomorrow unless there are objections.

mikofski, 2020-04-21:
Sorry all, no pressure! Let's not worry about NumFOCUS newsletter, and do what's best for pvlib and solar. The newsletter is just a "nice-to-have" and quite frankly, it's hard for me to really measure what NumFOCUS has done for pvlib specifically, while I agree they have done a lot for numerical and scientific python in general which we also benefit from indirectly. Thanks!

mikofski, 2020-04-22:
Thanks @CameronTStark ! Great work getting this out! 🎉 🚀

I just sent an email to announce the release in the April NumFocus newsletter. Are you planning to post announcements anywhere. There is a list of mail lists to post to in [the wiki Release Procedures: publicize release](https://github.com/pvlib/pvlib-python/wiki/Release-procedures#publicize-release):

- [x] email NF newsletter, currently [Walker Chabbott](walker@numfocus.org)
- [ ] the [pvlib-python Google group](https://groups.google.com/forum/#!forum/pvlib-python)
- [ ] [NumFOCUS Google group](https://groups.google.com/forum/#!forum/numfocus)
- [ ] [PyData Google group](https://groups.google.com/forum/#!forum/pydata)
- [ ] scipy-user@python.org Python mailing list
- [ ] numpy-discussion@python.org Python mailing list
- [ ] python-announce-list@python.org Python mailing list

After any announcements go out, should we can close this issue?

CameronTStark, 2020-04-22:
Sounds good. Thanks for the checklist! I was just wrapping up the conda version upload.

mikofski, 2020-04-22:
Let me know how I can help.

CameronTStark, 2020-04-23:
Just sent them all out as one bcc'd. I see the pvlib-python and NumFOCUS groups posting alright but PyData hasn't shown up.

Are the python.org ones email only or is there a forum I'm not seeing for them?

mikofski, 2020-04-23:
You must first sign up for all of these groups, forums, & mail lists
* PyData takes forever, I think it's moderated? But [it's up now](https://groups.google.com/forum/m/#!topic/pydata/gKOJ9lgVJ3Q)
* Yes the python.org are mail lists, but they have websites to view archives:
    - https://mail.python.org/mailman/listinfo/numpy-discussion
    - https://mail.python.org/mailman/listinfo/scipy-user
    - https://mail.python.org/archives/list/python-announce-list@python.org/latest

I can see the [Python-announce-list pvlib v0.7.2 posted](https://mail.python.org/archives/list/python-announce-list@python.org/thread/Q6J62TP7N4WAMN6FZEBAS4SR7QQQDPFR/). Numpy-discussion & SciPy-user still waiting. Sometimes they take a while too. [NumFOCUS Google group is posted](https://groups.google.com/forum/m/#!topic/numfocus/CfrQEX3buDI) and pvlib also up. Thanks!

CameronTStark, 2020-04-23:
Thanks for compiling all that @mikofski! I'll keep an eye out for the numpy and scipy lists since those are the only ones that I haven't seen go out yet.

I've just pushed a commit with a new "What's New" file and updated the Zenodo badge link on the Readme so everyone is ready to continue with v0.7.3. 

I've also moved the milestone for the unfinished v0.7.2 item to v0.7.3 and adjusted the milestone's due date from May 1st to June 22nd tentatively. 

I believe that wraps up v0.7.2. I'll leave this open for another day or so in case anything comes up to discuss.  

I will open up another issue for v0.7.3 like this one soon.

Thank you all for your support and great contributions!

cwhanse, 2020-04-23:
Closed via [v0.7.2](https://github.com/pvlib/pvlib-python/releases/tag/v0.7.2)

closed 2020-04-23 by cwhanse