pvlib/pvlib-python #867: Implement bilinear interpolation for efficiency matrix measurements

adriesse, 2020-01-24:
IEC 61853-3 recommends bi-linear interpolation as a method to use a matrix of efficiency measurements for calculating climate-specific energy ratings (CSER).  The description in the standard leaves some details open to interpretation, therefore some choices must be made in the implementation which are not part of the official standard.

CameronTStark, 2020-01-24:
@adriesse Thanks for your insight. I've labeled this as a question to highlight it as a discussion topic. For clarity, can you point to where in the code this would apply?

adriesse, 2020-01-24:
You'll see it when I post the code!  :)

adriesse, 2020-02-10:
Currently it seems like this should go into `pvsystem.py`, which is already rather large.  Is any reorganization around the corner that could influence this decision?

Or should we make `iec61853.py`.  There would be more functions in the pipeline for it, such as spectral mismatch calculations.

cwhanse, 2020-02-10:
This is a bit of a new creature at the zoo. Do you have in mind to implement the work flow from 61853 measurements to the climate specific energy rating? Some of the steps could be functions in other modules (e.g. `iam.py`), other steps don't have an analog in the library, like the efficiency interpolation or the spectral calculation.

If we had a code module that was only for PV module functions, that would be the place. `module.py` does not work - `pvmodule.py`? I'd name the function for the standard, `interpolate_efficiency_iec61853` or something like that.

According to the development plan, content in `pvsystem.py` will be moving to other modules as was done with `temperature.py` and `iam.py`. The idea is to more closely align the code with the steps in the modeling process. I'm considering `inverter.py` next.  After that the names don't seem to come naturally.

adriesse, 2020-02-10:
I could see room for some fitting function as well,  but still, `iec61853` would remain nice and small. If we called it `iec.py` it could accommodate more.

`pvmodule` would remain large and be susceptible to growing larger again, would it not?

cwhanse, 2020-02-11:
I'm -1 on naming a code module for a specific standard, since it is a detour from aligning code modules with modeling steps, and IEC61853 is basically laying out a specific path through the chain of model steps.

Interested in others views here - @mikofski? @btaylor?

Whatever name `pvmodule.py` takes, this new code module would have functions for the irradiance-to-DC conversion step, e.g., `calcparams_xxx`, `singlediode`, `pvwatts_dc`. It could also contain other functions to calculate quantities directly related to a module's DC output, e.g., fill factor, efficiency.

mikofski, 2020-02-11:
I'm with Cliff, I prefer `pvmodule.py` and then naming this specific 2-d interpolation method something like Cliff's suggestion: `interpolate_efficiency_iec61853()` but I have to ask, why would we re-implement 2-d interpolation, which is already covered by [SciPy `interp2d`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.interpolate.interp2d.html)? Is the IEC61853 method so specific that it needs it's own special implementation? It seems fragile unless absolutely necessary. One nice feature of using `scipy.interpolate.interp2d` is the `kind` option lets you select either linear or cubic.