pvlib/pvlib-python #875: Deprecation warning due to invalid escape sequences.

tirkarthi, 2020-02-01:
**Describe the bug**

Deprecation warning due to invalid escape sequences.

**To Reproduce**

```
find . -iname '*.py'  | xargs -P 4 -I{} python3.8 -Wall -m py_compile {}

./pvlib/iotools/ecmwf_macc.py:58: DeprecationWarning: invalid escape sequence \.
  """
./versioneer.py:485: DeprecationWarning: invalid escape sequence \s
  LONG_VERSION_PY['git'] = '''
```

mikofski, 2020-02-01:
Thanks! It's in the docstring, there's a backwards path separator, you can see in the documentation, it doesn't render correctly, easy fix

mikofski, 2020-02-01:
@tirkarthi seems okay now, can you please check again? Thx!
[image]

tirkarthi, 2020-02-01:
versioneer file also needs usage of raw strings prefix.

mikofski, 2020-02-02:
Versioneer is automatically generated, I think this issue has already been reported over there, [Warner/python-versioneer: flake8 complaint #201](https://github.com/warner/python-versioneer/issues/201)

PR #876 (2020-02-11): Fix backwards path sep
Fixes #875

 - [x] Closes #875
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Visually confirmed docs rendered correctly
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with ``  [... cut]
commit a18ee108 (2020-02-11): Fix backwards path sep (#876)

Fixes #875
commit b4486c16 (2020-02-13): remove needs_pandas decorator (#885)

* remove definition and applications of pandas_0_22 decorator

* remove needs_pandas_0_17

* Fix backwards path sep (#876)

Fixes #875

* readd needs_pandas_0_22 for test_irradiance.py in travis py35-min

* forecast compat with pandas 1.0, fix bug in Location tz handling (#879)

* fix location tz bug with datetime.timezone.utc

* update whatsnew

* issue in whatsnew

* fix whatsnew class typo

* remove definition and applications of pandas_0_22 decorator

* remove needs_pandas_0_17

* readd needs_pandas_0_22 for test_irradiance.py in travis py35-min
closed 2020-02-11 by wholmgren