pvlib/pvlib-python #877: test_forecast failures with pandas 1.0.0

cwhanse, 2020-02-03:
Pandas has released v1.0.0.  With pandas 1.0.0, `test_forecast.test_datetime` fails, as #872 shows in the test log.  It is not clear if the test itself is the problem: [these lines](https://github.com/pvlib/pvlib-python/blob/fb1ab533ab9a4b87a3161d577215ef3d8f12ed74/pvlib/tests/test_forecast.py#L117) do not set a timezone.

```
    start = datetime.now()
    end = start + timedelta(days=1)
```

**Should `start` and `end` be timezone-aware?**  The intent of the test is not clear to me.

**To Reproduce**
Steps to reproduce the behavior:

1. Install pandas 1.0.0
```
import pandas as pd
from datetime import datetime, timedelta
from pvlib.forecast import NAM

_latitude = 32.2
_longitude = -110.9
_tz = 'US/Arizona'
_start = pd.Timestamp.now(tz=_tz)
_end = _start + pd.Timedelta(days=1)

amodel = NAM()
start = datetime.now()
end = start + timedelta(days=1)
try:
    amodel.get_processed_data(_latitude, _longitude, start, end)
    print("Success with {}".format(pd.__version__))
except:
    print("Failed with {}".format(pd.__version__))
```

Now run with pandas < 1.0.0 (e.g., 0.22.0)

wholmgren, 2020-02-03:
The test ensures that the function works if you use a python datetime object instead of pandas datetime object. We should check that start/end are localized and, if not, raise `TypeError` before issuing the query.

cwhanse, 2020-02-03:
Want to correct this in conjunction with #878? Or I can do it. The test failures will repeat for every PR unless we pin pandas <= 1.0.0.

wholmgren, 2020-02-03:
Yes, I should have said I'll provide a fix as soon as I get out of conda hell. Apparently I should not have run `pip install pandas -U` in an environment with conda-provided numpy/netcdf/hdf5. Yay python packaging.

PR #879 (2020-02-12): forecast compat with pandas 1.0, fix bug in Location tz handling

 - [x] Closes #877
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link  [... cut]
closed 2020-02-12 by CameronTStark