pvlib/pvlib-python #880: access private _parse_pvgis_tmy_csv() function as read_pvgis_tmy_csv()

mikofski, 2020-02-04:
**Is your feature request related to a problem? Please describe.**
someone sent me a csv file they downloaded from pvgis, and I needed to parse it, so I had to call the private methods like this:

```python
In [18]: with (path_to_folder / 'pvgis_tmy_lat_lon_years.csv').open('rb') as f:
    ...:     pvgis_data = read_pvgis_tmy_csv(f)
```

**Describe the solution you'd like**
If I need this, others may also. I think a public method that takes either a string or a buffer could be useful? Something called `read_pvgis_tmy_csv()`

**Describe alternatives you've considered**
I was able to do it by just calling the private function and it worked, so that's an alternative also

**Additional context**
related to #845 and #849

cwhanse, 2020-02-05:
OK with me to refactor to the same pattern as #842 in particular the get / read / parse [trio](https://github.com/pvlib/pvlib-python/pull/842#issuecomment-570309454).

mikofski, 2020-02-23:
I just started in this, but didn't push it yet, sorry. Kinda important now to have a static/remoteless test that checks parsing, since all of the tests are decorated

PR #907 (2020-03-01): add read pvgis tmy

 - [x] Closes #880
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or [... cut]
commit d6b85c1f (2020-03-01): add read pvgis tmy (#907)

* add read_pvgis_tmy

* for files downloaded from pvgis tool
* update tests, api, and docs
* allow commit .csv to pvlib/data

* update what's new with read_pvgis_tmy

* closes #880
* test import from api vs. full namespace path

* py35 no likey fstring in read_pvgis_tmy

* don't use is for string comparison, ditto
* state returns, add see also

* use str in json.load in read_pvgis

* test read_pvgis_tmy_json, don't use binary buffer

* try parse_epw in read_pvgis_tmy first

* since attribute error only occurs for str/path

* more comments, space between pvgis outputf [... cut]
closed 2020-03-01 by wholmgren