pvlib/pvlib-python #881: Equation for pvlib.irradiance.aoi_projection

BaptisteDE, 2020-02-06:
Hi,
I am trying to use pvlib.irradiance.aoi_projection function to to compute beam radiation on a plan. The objective is to fill gaps in experimental campaign with data from CAMS service.

I got the amount of radiation (W/m²) using the aoi_projection  function and multiplying by CAMS Beam normal radiation.  But throughout the year, the results were not consistent with the measurement. 

Looking at the equation, I had questions :
assuming :
- Surface tilt angle : 0° to <=180° and 0° is plan facing upward
- Solar zenith : angle in degree from the ground
- Surface azimuth : angle in degree between 0° and 360°. 0° being north, 90° being east, etc.
- Solar azimuth : angle in degree between 0° and 360°. 0° being north, 90° being east, etc.

The actual equation is
_projection = (
        tools.cosd(surface_tilt) * tools.cosd(solar_zenith) +
        tools.sind(surface_tilt) * tools.sind(solar_zenith) *
        tools.cosd(solar_azimuth - surface_azimuth))_

I understand the azimuth projection, but not the zenith

[image]

Shouldn't the first part of the equation be ?
_tools.sind(surface_tilt) * tools.cosd(solar_zenith) - tools.cosd(surface_tilt) * tools.sind(solar_zenith)_

Thank you !

cwhanse, 2020-02-06:
> Solar zenith : angle in degree from the ground

Thanks for the question.  Solar elevation is the angle in degrees from the horizon. Solar zenith is the complement, the angle in degrees from vertical.

BaptisteDE, 2020-02-06:
Thank you very much for your quick answer. Indeed I was not looking at the right angle. All my apologies for this question.

closed 2020-02-06 by BaptisteDE