pvlib/pvlib-python #884: ModelChain with PVSystem No aoi

japita-se, 2020-02-07:
I created a simple script to test the ModelChain:

```
# Access weather data
import pandas as pd
import matplotlib.pyplot as plt
import datetime
import numpy as np
# import pvlib forecast models
from pvlib.forecast import GFS, NAM, NDFD, HRRR, RAP
from pvlib import solarposition

# specify location (Napoli, IT)
latitude, longitude, tz = 40.863, 14.2767, 'Europe/Rome'

# specify time range: 2 giorni
start = pd.Timestamp(datetime.date.today(), tz=tz)
end = start + pd.Timedelta(days=1)

tilt=30

irrad_vars = ['ghi', 'dni', 'dhi']
# model a simple PV plan
from pvlib.pvsystem import PVSystem as pv, snlinverter, retrieve_sam
from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS
from pvlib.modelchain import ModelChain
from pvlib.tracking import SingleAxisTracker
 
# Import the GFS module
model = GFS()

# modules
sandia_modules = retrieve_sam('cecmod')
module = sandia_modules['A10Green_Technology_A10J_S72_175']

#inverters
inverters = retrieve_sam('sandiainverter')
inverter =inverters.ABB__MICRO_0_25_I_OUTD_US_240__240V_
# temperature model
temperature_model_parameters = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_glass']
# PV System ca 3kW
system = pv(surface_tilt=tilt,module_parameters=module, inverter_parameters=inverter,temperature_model_parameters=temperature_model_parameters,modules_per_string=5, strings_per_inverter=2)
 
# ModelChain
# wheather data
data = model.get_processed_data(latitude, longitude, start, end)
resampled_data = data.resample('15min').interpolate()
mc = ModelChain(system, model.location)
mc.run_model(resampled_data)
mc.ac.fillna(0)
```

I get:

`ValueError: could not infer AOI model from system.module_parameters. Check that the system.module_parameters contain parameters for the physical, aoi, ashrae or martin_ruiz model; explicitly set the model with the aoi_model kwarg; or set aoi_model="no_loss".`

This happens only if I select pvlib.pvsystem as system. If I select SingleAxisTracker, no problem
Is it a bug?

cwhanse, 2020-02-07:
A bug, at least. You should get the same behavior with either `PVSystem` or `SingleAxisTracker`. Can you post the pvlib version and the line of code that calls `ModelChain` with `SingleAxisTracker` as the PV system?

The AOI model must be specified with `ModelChain(..., aoi_model=<model_name>,...)`, or the AOI model parameters must be added to the `module_parameters` dict.  If a model name is used, default parameters are provided. The SAM module parameter database does not include parameters for the AOI models.

japita-se, 2020-02-10:
Hi cwhanse! Thank you for reply.
The pvlib is the latest 0.7.1.
For SingleAxisTracker, the call is as simple as:

`system = SingleAxisTracker(module_parameters=module, inverter_parameters=inverter, temperature_model_parameters=temperature_model_parameters, modules_per_string=15, strings_per_inverter=300)`

However, by looking at  this example
[simple example](https://pvlib-python.readthedocs.io/en/stable/modelchain.html#a-simple-modelchain-example)
it seems that the SAM modules do include the aoi_model, don't they?

cwhanse, 2020-02-10:
> However, by looking at  this example
> simple example
> it seems that the SAM modules do include the aoi_model, don't they?

That example is using module parameters for the Sandia module model `retrieve_sam('SandiaMod')` not from the SAM list of parameters for the CEC module model which would be `retrieve_sam('cecmod')`. The file with Sandia module parameters does contain values for one of the IAM models.

cwhanse, 2020-02-10:
I get the same `ValueError` using either a fixed or tracked system. Here's my code with the GFS etc. steps stripped out, since the weather data is not needed to define the IAM model in the ModelChain instance.  What's different from your code where you don't get the `ValueError` when using a tracked system?

```
import pandas as pd
import datetime
from pvlib import location

# specify location (Napoli, IT)
latitude, longitude, tz = 40.863, 14.2767, 'Europe/Rome'
location = location.Location(latitude=latitude, longitude=longitude, tz=tz)

# specify time range: 2 giorni
start = pd.Timestamp(datetime.date.today(), tz=tz)
end = start + pd.Timedelta(days=1)

tilt=30

# model a simple PV plan
from pvlib.pvsystem import PVSystem as pv, retrieve_sam
from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS
from pvlib.modelchain import ModelChain
from pvlib.tracking import SingleAxisTracker
 
# Import the GFS module
#model = GFS()

# modules
sandia_modules = retrieve_sam('cecmod')
module = sandia_modules['A10Green_Technology_A10J_S72_175']

#inverters
inverters = retrieve_sam('sandiainverter')
inverter =inverters.ABB__MICRO_0_25_I_OUTD_US_240__240V_
# temperature model
temperature_model_parameters = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_glass']
# PV System ca 3kW
system1 = pv(surface_tilt=tilt,module_parameters=module, inverter_parameters=inverter,temperature_model_parameters=temperature_model_parameters,modules_per_string=5, strings_per_inverter=2)

# tracked system
system2 = SingleAxisTracker(module_parameters=module, inverter_parameters=inverter, temperature_model_parameters=temperature_model_parameters, modules_per_string=15, strings_per_inverter=300)

# ModelChain

#mc1 = ModelChain(system1, location)
mc2 = ModelChain(system2, location)
```

closed 2020-02-13 by wholmgren