pvlib/pvlib-python #901: switch from pandas.util.testing to pandas.testing

wholmgren, 2020-02-21:
`pandas.util.testing` is deprecated and we're supposed to switch to the functions in the public API at `pandas.testing` instead. 

I'm not sure what the pandas version compatibility is for this. If there's a problem, we could use `try/except` statements to handle earlier versions of pandas. Or we could just do nothing until the min pandas we support is high enough.

CameronTStark, 2020-02-21:
I didn't find a time horizon when [`pandas.util.testing`](https://github.com/pandas-dev/pandas/blob/master/pandas/util/testing.py) would disappear. That module is currently executing `from [pandas._testing](https://github.com/pandas-dev/pandas/blob/master/pandas/util/testing.py) import *` so changes to `pandas._testing` will affect us eventually. My guess is probably not until a 1.1.0 release at the earliest. 

At this point I'm +1 for waiting and aiming this for our v0.8.0 milestone where we can bump pandas-min to this level.

wholmgren, 2020-02-21:
Looks like it was added in pandas 0.20. https://github.com/pandas-dev/pandas/pull/16003

I think we're talking about going up to pandas 0.22 in pvlib 0.8, so let's make this a pvlib 0.8 milestone.

kanderso-nrel, 2020-07-09:
I think this will only take effect in pandas 2.0, according to their versioning policy: https://pandas.pydata.org/docs/whatsnew/v1.0.0.html#new-deprecation-policy

That doesn't mean we have to wait until then to make the switch.  @wholmgren are you imagining a straightforward find and replace for this, or maybe something like importing the pandas functions in conftest and then doing `from conftest import assert_series_equal` and such in the test files?

wholmgren, 2020-07-09:
> That doesn't mean we have to wait until then to make the switch. 

If we're bumping the minimum pandas in 0.8 then can we just go ahead and do the switch now?

> @wholmgren are you imagining a straightforward find and replace for this, or maybe something like importing the pandas functions in conftest and then doing from conftest import assert_series_equal and such in the test files?

I was imagining a find/replace. I don't see any advantage to adding it to conftest but let me know if I'm missing something.

kanderso-nrel, 2020-07-09:
> If we're bumping the minimum pandas in 0.8 then can we just go ahead and do the switch now?

Sure.

> I was imagining a find/replace. I don't see any advantage to adding it to conftest but let me know if I'm missing something.

Only advantage I see is that if pandas changes it again, we would only update one file instead of many.  Seems unlikely.  Find/replace is fine with me.

PR #1003 (2020-07-19): Bump minimum pandas version to 0.22.0, switch from pd.util.testing to pd.testing
 - [x] Closes #901
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - ~Tests added~
 - ~Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.~
 - ~Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includ [... cut]
closed 2020-07-19 by wholmgren