pvlib/pvlib-python #911: Sphinx emits lots of warnings when building docs

kanderso-nrel, 2020-02-25:
Sphinx currently emits a few types of warnings when building the docs, none are really a big deal but it clutters the output and makes #909 less effective.  I'll open a PR shortly to address:

Docutils warning
----------------------
```
/usr/share/miniconda/envs/test_env/lib/python3.8/site-packages/sphinx/util/nodes.py:94: FutureWarning: 
   The iterable returned by Node.traverse()
   will become an iterator instead of a list in Docutils > 0.16.
```

I'm not 100% sure what this one is, but I assume it is related to the newest version of docutils deprecating something and sphinx hasn't caught up yet.  [example](https://dev.azure.com/solararbiter/pvlib%20python/_build/results?buildId=2804&view=logs&j=7e620c85-24a8-5ffa-8b1f-642bc9b1fc36&t=2febcae8-7f6a-51bb-45aa-d6e72c9ea7e2&l=128)

I'll play around with pinning the docutils version and see if that fixes it. 

Improper RST in gallery example
------------------------------------------

The I-V curve example generates `WARNING: Title underline too short.` because the page title got updated without extending the underline (my bad).  Easy fix. [link](https://dev.azure.com/solararbiter/pvlib%20python/_build/results?buildId=2804&view=logs&j=7e620c85-24a8-5ffa-8b1f-642bc9b1fc36&t=2febcae8-7f6a-51bb-45aa-d6e72c9ea7e2&l=187) 

First-line references
-------------------------

The first lines of docstrings get orphaned when they are copied into the API Reference docs page and Sphinx emits a bunch of warnings about it: [example](https://dev.azure.com/solararbiter/pvlib%20python/_build/results?buildId=2804&view=logs&j=7e620c85-24a8-5ffa-8b1f-642bc9b1fc36&t=2febcae8-7f6a-51bb-45aa-d6e72c9ea7e2&l=154)

There were a few votes in #909 to remove references from first lines of docstrings to prevent these warnings.  If nobody objects, I'll move references into the second part of the docstring, for instance:

Current:
```
def nrel_earthsun_distance(time, how='numpy', delta_t=67.0, numthreads=4):
    """
    Calculates the distance from the earth to the sun using the
    NREL SPA algorithm described in [1]_.

    Parameters
    ----------
    time : pandas.DatetimeIndex
```

New:
```
def nrel_earthsun_distance(time, how='numpy', delta_t=67.0, numthreads=4):
    """
    Calculates the distance from the earth to the sun using the
    NREL SPA algorithm.

    The details of the NREL SPA algorithm are described in [1]_.

    Parameters
    ----------
    time : pandas.DatetimeIndex
```

This is related to #837 but that's a much bigger job and I'd rather not tackle it right now.

mikofski, 2020-02-25:
FYI: I fixed the 1st line reference issue for `losses.py` in #910 

https://github.com/pvlib/pvlib-python/blob/64ec94d811a6608842151b6b519f845f7ea9613e/pvlib/losses.py#L16-L21

https://github.com/pvlib/pvlib-python/blob/64ec94d811a6608842151b6b519f845f7ea9613e/pvlib/losses.py#L103-L111

and I also noticed that they show up for gallery examples in the tooltip popup, so I removed it from the Kimber soiling example here too:

https://github.com/pvlib/pvlib-python/blob/64ec94d811a6608842151b6b519f845f7ea9613e/docs/examples/plot_greensboro_kimber_soiling.py#L1-L30

kanderso-nrel, 2020-02-25:
Thanks for the heads up!  I'll leave those alone/merge them in from master when #910 is merged.

wholmgren, 2020-02-25:
Thanks!

> I assume it is related to the newest version of docutils deprecating something and sphinx hasn't caught up yet. example

Should we bump sphinx instead of (or in addition to) pinning docutils?

> The first lines of docstrings get orphaned when they are copied into the API Reference docs page and Sphinx emits a bunch of warnings about it:

Is it the first line that matters or the first sentence or the first paragraph? #912 seems to take a few different approaches.

mikofski, 2020-02-25:
>Should we bump sphinx instead of (or in addition to) pinning docutils?

+1

I just noticed yesterday that Sphinx is pinned at v1.8.5, which IMO is pretty old. I don't think this a usability concern because only for builders would care about this, so probably zero users would be affected by updating Sphinx to the newest version which is like 2.something I think.

I think it's okay to pin Sphinx and docutils after we confirm they are rendered to our satisfaction.

kanderso-nrel, 2020-02-25:
Sorry for delayed reply.  Pinning sphinx at 1.8.5 was decided in https://github.com/pvlib/pvlib-python/pull/833#discussion_r356951197 because sphinx 2.0+ (at the time) rendered references poorly.  I haven't tested it since then, so it could be that the most recent sphinx has improved reference formatting.  I'll play around with that tonight.  

> Is it the first line that matters or the first sentence or the first paragraph? #912 seems to take a few different approaches.

At first I thought that it was the first paragraph that mattered, but I realized halfway through that autodoc is only pulling the first English sentence (or at least something that looks like the first sentence).  So I think second sentence, first paragraph is fine for references.

wholmgren, 2020-02-25:
Thanks for the clarification. I guess the paragraph/sentence distinction wouldn't matter if we decided to follow [pep 257](https://www.python.org/dev/peps/pep-0257/#multi-line-docstrings).

kanderso-nrel, 2020-02-26:
Running with the most recent sphinx (2.4.3) and docutils (0.16) still produces the reference format seen in #833.  After trying it out for a bit, I think the CSS override necessary to get it looking nice again would be pretty unpleasant and I wouldn't want to maintain it.  However, sphinx 2.0+ does allow use of the original HTML4 writer use in sphinx 1.8.5, which produces good output with the newest sphinx/docutils.  The docutils warning is gone, but another has popped up:

```
/home/kevin/anaconda3/lib/python3.7/site-packages/sphinx_rtd_theme/search.html:20: RemovedInSphinx30Warning: To modify script_files in the theme is deprecated. Please insert a <script> tag directly in your theme instead.
  {{ super() }}
```

It's fixed as of https://github.com/readthedocs/sphinx_rtd_theme/pull/728, but unfortunately there hasn't been a release since the merge (https://github.com/readthedocs/sphinx_rtd_theme/issues/739) so isn't available on pypi.  I think this warning doesn't break the build, even with `-W`, so the only downside is extra clutter in the output.  So to summarize, these are the three options that give good output:
- Option 1 (current): sphinx==1.8.5, docutils==0.16: docutils warning
- Option 2 (#912 as of https://github.com/pvlib/pvlib-python/pull/912/commits/6098122882832f9a016e1343b3eb6e817639df3d): sphinx==1.8.5, docutils==0.15.2: no warning
- Option 3: sphinx==2.4.3, docutils==0.16, html4_writer=True: non-breaking warning

I don't really have a preference for Option 2 vs 3.  Curious what others think.

mikofski, 2020-02-26:
_my vote_: **Option 2** (#912 as of 6098122): `sphinx==1.8.5`, `docutils==0.15.2`: no warning

wholmgren, 2020-02-26:
Yikes what a mess. Thanks for the research! I’m ok with either 2 or 3.

kanderso-nrel, 2020-02-26:
I'd say let's do option 2 for now and revisit after sphinx_rtd_theme cuts a new release.  From reading through their issue log, it seems like the next release is focused on improving support for sphinx 2+ anyway so it should be a good time to bump our sphinx version.

PR #912 (2020-03-03): Fix most sphinx warnings
 - [x] Closes #911 
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [x] New code is fully documented. In [... cut]
closed 2020-03-03 by wholmgren