pvlib/pvlib-python #916: Separating changes to the CI in "What's New"

CameronTStark, 2020-02-26:
**Is your feature request related to a problem? Please describe.**
Changes that are being made to the CI are mixed into bug fixes for the functional code base. It would be beneficial to users and maintainers to separate these.

**Describe the solution you'd like**
To make a CI section in "What's new" and move recent changes there. 

**Describe alternatives you've considered**
Leaving it as is.

cwhanse, 2020-02-27:
I agree that we shouldn't mix notices about CI changes with notices about code changes. But do we want to have CI changes documented in Whats New?  I'm not sure what is typical here.

wholmgren, 2020-02-27:
I think most of the non-trivial changes to CI have gone into the Testing section, which seems appropriate to me. Am I missing something?

CameronTStark, 2020-02-28:
The Testing section seems like a better fit. I'll move my the CI items there and close this issue. Thanks for your feedback!

closed 2020-02-28 by CameronTStark