pvlib/pvlib-python #920: ValueError:  Big-Endian buffer not supported on little-endian compiler

kahemker, 2020-02-29:
**Describe the bug**
Sometimes when using the `process_data` function in `forecast.py` (usually by invoking `get_processed_data`), the script would crash with the error:

`ValueError:  Big-Endian buffer not supported on little-endian compiler`

I found the bug by using `forecast.get_processed_data()` function. I can pull forecast data consistently without errors using the only the `get_data()` functions. 

`ValueError: Big-endian buffer not supported on little-endian compiler`

**To Reproduce**
I creating this bug on Windows 10-64 Bit OS with Python 3.8.2.  You can follow the script written in this [StackOverflow ](https://stackoverflow.com/questions/60161759/valueerror-big-endian-buffer-not-supported-on-little-endian-compiler/60466935#60466935)post and occasionally you should see this error. 

**Expected behavior**
Data retrieved should not create runtime errors during processing

**Versions:**
 - `pvlib.0.7.1` 
 - `pandas.1.0.1`
 - `python: 3.8.2`

**Additional context**
I believe I can fix this bug.  I pushed my modified `forecast.py` [here](https://github.com/kahemker/pvlib-python/tree/big-endian-buffer-bug).

wholmgren, 2020-03-02:
Thanks for digging into this @kahemker. Can you also post the versions of netcdf and hdf5 that you're using? 

@dopplershift, do you know anything about the byte-order behavior for netcdf subset on THREDDS? This issue and the SO post linked to above suggest that the returned data is sometimes big-endian and sometimes little-endian. This causes a problem when trying to convert to a `pandas.Series`. I thought netcdf was supposed to make all this just work. Thanks in advance!

kahemker, 2020-03-02:
I am using netcdf4 version 1.5.3 and I am not using hdf5.  It does not appear in the setup.py list of dependencies for pv-lib (required or optional).

[Here is a pastebin ](https://pastebin.com/1TjbdiDJ)that has my entire conda environment.yml file.

dopplershift, 2020-03-02:
I'm pretty sure netCDF is supposed to be dealing with this. @WardF can you shed any light on netCDF endian handling?

Can someone point to the data in question?

WardF, 2020-03-02:
NetCDF handles endianness under the hood transparently; however, we are dealing with an issue currently (see https://github.com/Unidata/netcdf-c/issues/1338) where there appears to be a problem.  We recently got some Big Endian hardware in-house, to debug this for our upcoming 4.7.4 release.

PR #921 (2020-07-17): Fix big-endian buffer not supported on little-endian compiler bug

 - [ ] Closes #920 
 - [ ] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes lin [... cut]
closed 2020-07-17 by wholmgren