pvlib/pvlib-python #924: Add function to retrieve example datasets

kanderso-nrel, 2020-03-02:
It would be nice to have a clean way for gallery examples to get access to the test data files.  See https://github.com/pvlib/pvlib-python/pull/860#discussion_r379847062 

For a function like `load_dataset('greensboro-tmy')`:

Pros:
- No need to monkey around with filepaths, especially ones that aren't really meant to be public
- Associating files with keys means we can move and rename test data files without it being a breaking change
- Simplifies example code

Cons:
- The data files are in several formats (csv, json, nc, h5 etc), so this function would either have to know the appropriate reading method for each file (complicated) or just return a file handle and let the user parse the contents (less useful).

cwhanse, 2020-03-02:
+1 to having the function. I think a useful scope is to return the full path to the file. Reading file content and providing output in various formats seems like a large bite to chew.

kanderso-nrel, 2020-03-02:
Any ideas for what the function should be named if it returns the file path?  I'm having trouble coming up with a verb that doesn't seem misleading.  Maybe just `dataset()`? 

`df = pd.read_csv(dataset('greensboro-tmy'))`

`with open(dataset('greensboro-tmy')) as f:`

mikofski, 2020-03-02:
>Con: The data files are in several formats (csv, json, nc, h5 etc)

To me this a "Pro" for having a dedicated data reader, to make it easier for users, if it's a PITA for us, it will be a blocker for them I think

wholmgren, 2020-03-02:
I would be happy if this

https://github.com/pvlib/pvlib-python/blob/c9929e85986a3368ed370203ef8699617c0cdc61/docs/examples/plot_greensboro_kimber_soiling.py#L39-L43

looked like

```python
# get full path to the data file
file_path = dataset('greensboro-tmy')

# parse TMY3 data
greensboro, _ = read_tmy3(file_path, coerce_year=1990)
```

I don't think the broader scope is feasible. To be clear, this is just something for the tests/examples - not for anything else.

metpy has a `get_test_data` function with the same idea, but a different implementation because it uses a caching back end that I think we should avoid.

example: https://unidata.github.io/MetPy/latest/examples/XArray_Projections.html#sphx-glr-examples-xarray-projections-py

kanderso-nrel, 2020-03-17:
I just stumbled across some functions in pkg_resources that might be relevant to this issue: https://setuptools.readthedocs.io/en/latest/pkg_resources.html#resource-extraction

echedey-ls, 2023-03-13:
I just stumbled upon this issue looking for ideas for GSoC.
@kanderso-nrel , just a quick note on that might-be-relevant package:
From my experience both with PVLIB and SciencePlots, sticking to `__path__[0]` (which is already in use in PVLIB) is more than enough, and I believe (or want to) it is exempt from errors. At least, in SciencePlots we haven't recorded any issues regarding that after the breaking changes in the v2.0.0 release.
About `pkg_resources`, it's use is deprecated in favor of other packages. See [this attention directive at pkg_resources](https://github.com/pypa/setuptools/blob/be6c0218bcba78dbd4ea0b5a8bb9acd5d5306240/docs/pkg_resources.rst?plain=1#L13).