pvlib/pvlib-python #925: Adding the Loss Factors Model and Mechanistic Performance Models to PVLIB

steve-ransome, 2020-03-02:
I'll be adding the Loss Factors Models and Mechanistic Performance Models to PVLIB, . 

See http://www.steveransome.com/PUBS/1906_PVSC46_Chicago_Ransome.pdf for more details on what they do

I hope it should be ready by the conference in Salt Lake in May

I've uploaded a couple of graphs and some sample code and Initially I'm looking for feedback and comments as I will need to enhance the present naming convention

With the LFM I have been using prefix and suffix letters to describe the type of measurement and any corrections. For example this Python 3 code shows the naming I have been using and does normalised LFM and temperature correction.
I have just shown nIsc and nVoc for clarity but will upload code for all variables soon plus graphs and MPM fits.

Once I have this in a correct form I can add more of features

[image]
[image]
[MLFM_SRCL_200302.py.txt](https://github.com/pvlib/pvlib-python/files/4278331/MLFM_SRCL_200302.py.txt)

Any feedback is welcome
Steve

cwhanse, 2020-03-03:
@steve-ransome looking forward to hosting this function. Have you thought about the function's signature?  What are the inputs, what is output?

I should look at your PVSC paper again, refresh my memory, but Sandia is blocking your website. Would you mind sending it to me directly?

steve-ransome, 2020-03-03:
Thanks Cliff.

I've been writing Python code for myself and clients for a few years. I am not yet familiar with these collaborative efforts although I am learning. The python I added is just simple, sample code,to calculate some of the LFM coefficients and show my naming convention which allows variables  to be identified as "reference, measured or normalised" and also "uncorrected  or  temperature corrected" (e.g. "nIsc_T") so need an idea if this is OK to use as the naming convention document just allows names like i_sc and v_oc.

Basically the LFM function will transform measured values from an IV curve to normalised values

Inputs
IV curve : (mIsc, mRsc, mmp, mVmp, mRoc, mVoc ...)
Reference module STC values : (rIsc, rImp, rVmp, rVoc , rIsc_Alpha etc. ...)
Weather data : (Irradiance Gi, module temperature Tm, Windspeed WS ...)

Normalised Outputs
LFM values  : (nIsc_U, nRsc, nImp, nVmp, nRoc, nVoc_U and temperature corrected  _T) if desired

The MPM function fits up to n=6 normalised coefficients to predict a normalised LFM output such as nIdc or nVoc

nLFM = fn ( Gi, Tmod, WS, C1 .. Cn)

I've been reading up on some of the coding examples such as the 1-diode model to get an idea of the sort of style used for comments, layout etc. and hope to follow that.  Maybe if I follow https://www.python.org/dev/peps/pep-0362/ I can get you code that will be closer to what's needed,

Here are  my Chicago files uploaded - please tell me if you can get them

[1906_PVSC46_Chicago_Ransome_ppt_Presented.pdf](https://github.com/pvlib/pvlib-python/files/4282478/1906_PVSC46_Chicago_Ransome_ppt_Presented.pdf)

[1906_PVSC46_Chicago_Ransome.pdf](https://github.com/pvlib/pvlib-python/files/4282385/1906_PVSC46_Chicago_Ransome.pdf)

Is anyone else blocked from www.steveransome,com ? 

Does Sandia give a reason why my website is blocked? Perhaps you could ask them to unblock it, or give me a reason why not

I appreciate the amount of work you guys have done already over the years and will try to contribute as best as I can - even though I don't understand all the stuff mentioned in github.

cwhanse, 2020-03-03:
> allows variables  to be identified as "reference, measured or normalised" and also "uncorrected
> or  temperature corrected" (e.g. "nIsc_T") so need an idea if this is OK to use as the naming 
> convention document just allows names like i_sc and v_oc.

We can work through naming conventions for these new quantities with the pull request. It really helps to see the variables in context.

> I've been reading up on some of the coding examples such as the 1-diode model to get an idea of the sort of style used for comments, layout etc. and hope to follow that.  Maybe if I follow https://www.python.org/dev/peps/pep-0362/ I can get you code that will be closer to what's needed,

I don't think we are using PEP-362. By "signature" I only meant a list of inputs and outputs.

I can get the files from here, thank you. www.steveransome.com is not blocked, but the links to the hosted files are blocked here.  Hard to say why without a dozen phone calls.

steve-ransome, 2020-03-03:
Thanks. Maybe I can send some code over in a few days where I've tried to follow everything and made clear when I can't achieve something (like naming) as it is.

As it's Python it should be clear what it is I am trying to do but I do add comments too. You can see the basics from the first py file I sent

steve-ransome, 2020-08-13:
I presented a virtual talk at "August 5, 2020 PVPMC Webinar on PV Performance Modeling Methods"
https://pvpmc.sandia.gov/resources-and-events/events/2020-pvpmc-webinars/ 

"How to use the Loss Factors and Mechanistic Performance Models effectively with PVPMC/PVLIB"
https://pvpmc.sandia.gov/download/7879/

After receiving some positive feedback I'll soon be to be able to upload code as described in slides 15 and 16. 

What's the best way to go about this? I've followed as many of the hints and tips as I can find for coding, documentation and naming but I'm sure there will be changes needed
.
There'll be an ipynb file, a py lib file, some default measurement data, reference measurements and it will generate the 5 graphs in slide 15. Instructions are provided for you to use your own measurement data.

I am not familiar with the procedures you've already been using and welcome suggestions and links.

[image]

mikofski, 2020-08-13:
Hi Steve,

This is very exciting and welcome news! Thank you for offering your contributions to pvlib!

The best source of info would be the [contributing section of the pvlib documentation](https://pvlib-python.readthedocs.io/en/latest/contributing.html#contributing). Here are the salient points:

1. pvlib python uses Git & GitHub to manage contributions. There are other methods but this will give you the most control to ease integration of your code. Therefore you will need to have:
    * Registered for a [GitHub](https://gitHub.com) account
    * Forked the pvlib python repo to your personal GitHub account. Click the fork button on the [pvlib python GitHub page](https://github.com/pvlib/pvlib-python)
    * Installed [Git](https://git-scm.com/) on your computer. If you prefer a graphic user interface (*), after installing Git, get [tortoiseGit](https://tortoisegit.org/) which adds icon overlays and subcontext menus to Windows explorer

2. The easiest way to integrate your code into pvlib python is on *your* machine first! This is done by "cloning" the pvlib Git repository to your computer. If you installed tortoise the browse to the location where you want to work on your development version of pvlib, right click the folder and select "clone" then enter the URL to your fork of the GitHub pvlib python repo, *eg*: https://github.com/<steve-ransom?>/pvlib-python, then enter your GitHub credentials when prompted (**)

3. pvlib python uses a strategy called feature branching to integrate enhancements like the loss factor model. If using tortoise, navigate the the Git repo you just cloned, right click anywhere and select "checkout new branch". Them enter a descriptive branch name like "loss-factor-model"

4. Now integrate your code into the branch in your computer. After you've made the changes, you need to "commit" them to the repository. Right click again and select "commit". You should see that there are "untracked" changes. Click the files you want to add, add a descriptive comment and click "commit".

5. Next you need to make your changes known to GitHub by pushing them to your fork. Using tortoise, tight click and select "push". In the field select "origin" and the name of your branch: "loss-factor-model", then click okay. Enter your GitHub credentials again if prompted.

6. The last step is to let the maintainers know you want to integrate your changes on GitHub by creating a "pull request"online. Go to your online fork of pvlib python and you should see a message noting that GitHub observed your changes and offering to open a pull request from you to the main pvlib python repo. Do it.

Here are some more references to help you get started:
* https://bids.github.io/dats/posts/2018-09-10-github-oss.html
* https://bids.github.io/dats/posts/2017-10-04-github-oss-f17.html#using-github-for-open-source-projects
* http://poquitopicante.blogspot.com/2016/10/winning-workflow.html
* https://mikofski.github.io/BasicGitWorkFlow/

mikofski, 2020-08-13:
Here are the additional notes where I left asterisks
* (*) There are other graphic user interface for Git, notably [GitHub desktop](https://desktop.github.com/) & [Atlassian sourcetree](https://www.sourcetreeapp.com/), but I personal recommend just using the command line. The [Pro Git book](https://git-scm.com/doc) is IMHO very good. [GitHub](https://docs.github.com/en) & [Bitbucket](https://www.atlassian.com/git/tutorials) help can also be useful.
* (**) Another scheme for cloning your repo uses SSH keys instead of HTTPS. This requires you to create SSH private & public keys on your computer first, and then copying the public key online in your personal GitHub account settings

steve-ransome, 2020-08-13:
Thanks @mikofski , I'll try that over the weekend, there's a lot to take in!

steve-ransome, 2020-08-18:
Thanks @mikofski , I've been through all of your documentation but am still struggling to get this set up properly. I've used tortoisegit 

I've tried to add my mlfm files in a directory pvlib_python\pvlib\mlfm. There are a notebook (mlfm_pvpmc1.ipynb), library (mlfm_lib_1.py) then reference (ref, test and matrix.csv) and measurement files (*.csv) and a graphs directory where it automatically saves generated .png files. The ipynb and .py files have a lot of documentation. I've just put in the first couple of graphs and once it's working can update and add more.

Please can anyone help let me know if this is OK or what else I should do.

steve-ransome, 2021-11-01:
Thanks for the comments. I'm happy to maintain the python code, however I would need help on any github, testing and decisions on library.

I've had several useful comments from Cliff Hansen already and am trying to follow them all where possible.

Here's a zip file with all the relevant mlfm code and 3 reference files, the html files show the output from running each file.
[mlfm_0_4.zip](https://github.com/pvlib/pvlib-python/files/7454451/mlfm_0_4.zip)

Cliff suggested having different pandas data files e.g. meas[i_sc .. v_oc], norm[i_sc .. v_oc], stack[i_sc .. v_oc] rather than the way I had done it adding new columns to the same frame e.g. data[meas_i_sc .. meas_v_oc, norm_i_sc .. norm_v_oc, stack_i_sc .. stack_v_oc]. I can see the positives with doing this but I have problems when doing joins as I then have to rename several i_sc columns to have the correct prefix e.g. meas_ and norm_.

Thanks Anton for your comment, that's basically what I am trying to do to keep as little mlfm specific code as possible in the py file, I can do simple graphics such as figs 6,7 and 8 by defining them in the ipynb tutorial, however figs 3 and 5 are far more complex and I wonder if there could be a library for more complicated graphics.

I welcome all comments and suggestions!

adriesse, 2021-11-01:
I agree that appending more and more columns to a DataFrame may not be the best way.  If you do need to do joins, there are rsuffix and lsuffix arguments that will help to avoid column name conflicts.

For overall code structure/organization, perhaps you can split your current py file into two: one with pure model functions (no matplotlib and no printing), and one with the rest.  Perhaps that second file can then just be considered part of the example/tutorial.

steve-ransome, 2021-11-01:
Thanks Anton. Will try to adopt all your suggestions. I had forgotten about the suffix options, they should work on the joins.

steve-ransome, 2021-11-05:
I've rewritten the mlfm code now so it follows the helpful suggestions made.
I separated the py functions into 

**mlfm_lib_code_0_41.py** 

```
def mlfm_meas_to_norm(dmeas, ref, qty_mlfm_vars):
    return dnorm

def mlfm_6(dmeas, c_1, c_2, c_3, c_4, c_5, c_6):
    return mlfm_6

def mlfm_norm_to_stack(dmeas, dnorm, ref, qty_mlfm_vars):
    return dstack

def mlfm_fit(dmeas, dnorm, mlfm_sel): 
    return(dnorm, cc, ee, coeffs, errs)

```

**mlfm_lib_graphs_0_41.py**

```
def plot_mlfm_scatter(dmeas, dnorm, mlfm_file_name, qty_mlfm_vars):

def plot_mlfm_stack(dmeas, dnorm, dstack, ref, mlfm_file_name, qty_mlfm_vars,
    xaxis_labels=12, is_i_sc_self_ref=False, is_v_oc_temp_module_corr=True)
```

There are now three different dataframes : dmeas, dnorm and dstack

There are three different measurement files and one for testing, as I've taken out the print statements this is selected by setting meas_file

```
# select one of the following meas files 
meas_file = 0

if meas_file == -1:
    mlfm_meas_file = 't1_041.csv'
elif meas_file == 0:
    mlfm_meas_file = 'g78_T16_Xall_F10m_R900_041.csv'
elif meas_file == 1:
    mlfm_meas_file = 'n05667_Y13_R1k6_fClear_041.csv'
elif meas_file == 2:
    mlfm_meas_file = 'x19074001_iec61853_041.csv
```

This is the new version that replaces everything in version 04

[mlfm_0_41.zip](https://github.com/pvlib/pvlib-python/files/7484262/mlfm_0_41.zip)

I've included an html file showing how it looks running the 0Gantner data,

Again, any thoughts or comments welcome.
Steve

adriesse, 2021-11-11:
Time flies!  I think this is better for sure, but I think we need some more guidance from the maintainers here.  Is it close enough to have a PR?

cwhanse, 2021-11-11:
> Is it close enough to have a PR?

I think so. Python modules should be put into a new folder `pvlib.mlfm`.

steve-ransome, 2021-11-23:
What's the next stage to get this going? If I move the python modules to pvlib.mlfm can someone do a PR? I had problems last time I tried and would appreciate some help.

cwhanse, 2021-11-26:
@steve-ransome the next step is a PR so all maintainers can review. What problems did you have? Git has a learning kerb, for sure.

steve-ransome, 2021-11-29:
Thanks Cliff. I reported problems on 18 Aug 2020 further up this thread, I couldn't seem to understand what I was doing re github, tortoise git and the PR despite Mark's @mikofski instructions

cwhanse, 2021-12-02:
@steve-ransome my concern about creating the PR for you, is that someone will still need to use git to submit changes during the review. My workload is such that I don't think I can edit the PR to completion.

steve-ransome, 2021-12-03:
Thanks Cliff. I will contact you as requested.