pvlib/pvlib-python #927: cell temperature from backside of module temperature measurements

mikofski, 2020-03-03:
**Is your feature request related to a problem? Please describe.**
recently a question was posed, given backside module temperature, how to include them in the model chain? Looking at the temperature models available, there isn't one that takes module temperature as input and returns cell temperature. [SAPM has a formula for adjusting module temperature to obtain cell temperature](https://pvpmc.sandia.gov/modeling-steps/2-dc-module-iv/cell-temperature/sandia-cell-temperature-model/) based on irradiance and constant "delta-T" parameter that is characteristic of the module. However this formula is bundled into `pvlib.temperature.sapm_cell`as [Eq (2)](https://pvlib-python.readthedocs.io/en/latest/generated/pvlib.temperature.sapm_cell.html#equation-sapm2 ) and the function doesn't take module temperature as input

**Describe the solution you'd like**
add an optional kwarg to `pvlib.temperature.sapm_cell` that allows it to bypass the module temperature calculation.

https://github.com/pvlib/pvlib-python/blob/ae98bf94413ac1495c8fe3ed4ce985a3f74c3805/pvlib/temperature.py#L117

##### option 1:
add `module_temperature` as kwarg
```python
def sapm_cell(poa_global, temp_air, wind_speed, a, b, deltaT,
              irrad_ref=1000, module_temperature=None):
    if module_temperature is None:
        module_temperature = sapm_module(poa_global, temp_air, wind_speed,
    else:
        warning.warn('air temp and wind speed parameters ignored, overridden by module temp')
        return module_temperature + (poa_global / irrad_ref) * deltaT
```

##### option 2:
add new `pvlib.temperature.sapm_module_to_cell` func
```python
def sapm_module_to_cell(module_temperature, poa_global, deltaT, irrad_ref=1000):
    return module_temperature + (poa_global / irrad_ref) * deltaT
```

_note_: if option 2, then `pvlib.temperature.sapm_cell` could be updated to call `pvlib.temperature.sapm_module_to_cell`

**Describe alternatives you've considered**
* We could explain to users that they can _back out_ the conditions to the use `pvlib.temperature.sapm_cell`
* we could encourage users to formulate the equation themselves from the literature
* we could implement a 3rd SAPM function to perform this task seperately, call it `pvlib.temperature.sapm_module_to_cell`?

**Additional context**
related to [this post on the Google group](https://groups.google.com/d/msg/pvlib-python/CBThHDydaxs/oYuaKwJmAQAJ)

wholmgren, 2020-03-03:
-1 for a new kwarg, +1 for a new function

cwhanse, 2020-03-03:
+1 for `pvlib.temperature.sapm_cell_from_module` rather than `sapm_module_to_cell`, I'd prefer to keep the output earlier in the name string

PR #938 (2020-04-07): add temperature.sapm_cell_from_module
 - [x] Closes #927
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with ``  [... cut]
closed 2020-04-07 by CameronTStark