pvlib/pvlib-python #931: Rethinking pvlib.ivtools

cwhanse, 2020-03-09:
When I proposed `pvlib.ivtools` I had in mind only the use case of fitting single diode models for PV modules. I expect to see PRs soon with a function to fit the Sandia inverter model (to measured efficiency curves), and functions for the Loss Factor Model. These don't meet the narrow scope of the current  `ivtools`. There are likely other model fitting functions that could be of interest to pvlib users.

#708 refactors `ivtools` to separate functions into modules `sde` and `sdm` for single diode equation and model fitting, respectively. In hindsight, `ivtools` is too generic, since it doesn't communicate what the module does besides something with IV curves.

An alternate home for model fitting functions, including those in `ivtools`, might be  [pvanalytics](https://github.com/pvlib/pvanalytics).  But the inverter model fitting code doesn't seem natural there, either, since `pvanalytics` is primarily being designed for analysis of time-series data from PV systems.

Options:
- keep `pvlib.ivtools`, and stick `fit_inverter_snl` (or whatever it's named) into `pvlib.pvsystem`.
- add fitting functions to the code module containing the related model function, e.g., put `fit_inverter_snl` into `pvlib.pvsystem` or merge #886 and add `fit_inverter_snl` there.
- create a new module with a broader name than `ivtools` to contain all model fitting functions, divided up into appropriate modules.
- refer the two PRs to [pvanalytics](https://github.com/pvlib/pvanalytics)

Other ideas?

mikofski, 2020-03-09:
-1 on pvanalytics, IMO generating inverter and PV-module parameters should be the job of pvlib

My personal preference would be to split `pvsystem.py` out into `pvmodules.py` and `inverters.py`, then `fit_inverters*` and the inverter models would be in `inverters.py` and ditto for PV-modules. If that's too specific, then PV-modules and inverters could be in `components.py` which could be a catchall for PV-modules, inverters, trackers, batteries, cabling, xfmrs, etc. 
* IMO `pvsystem` is too long, #436
* `pvsystem` would still have `PVSystem` class and some system performance functions like `singlediode`, `max_power_point`, etc
* other component specific functions would be moved to their new homes, such as `calcparams*`, `retrieve_sam`, `snl_inverter`, `adrinverter`

```
*-*- pvsystem/
|  \
|   *- pvsystem.py
|   |
|   *- components.py
```

or another way to split them would be `pvsystem/performance.py` and `pvsystem/parameters.py`?

cwhanse, 2020-03-10:
I'd favor `pvmodule.py` and `inverter.py` over the (IMO) too-generic `component.py`, `parameters.py`, etc.

wholmgren, 2020-03-10:
+/- 0 on moving the fit code to `pvanalytics`. 

-1 on adding it to existing `pvsystem.py`.

+1 on breaking up `pvsystem.py` and uncertain on moving the fit functions into the relevant new modules.

-1 on a `pvsystem` subpackage at this point in time.

wholmgren, 2020-08-28:
ok to close this @cwhanse?

cwhanse, 2020-08-28:
yes

closed 2020-08-28 by cwhanse