pvlib/pvlib-python #932: Runtime warning in singleaxis tracker

HrushikeshBodas, 2020-03-10:
**Describe the bug**
~\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pvlib\tracking.py:555: RuntimeWarning: invalid value encountered in remainder
  surface_azimuth = surface_azimuth % 360

**To Reproduce**
I used this code :
`import pandas as pd
import numpy as np
import pvlib
azimuth = pd.read_csv("Azimuth.csv")
zenith = pd.read_csv("solar2.csv")
Plib = pvlib.tracking.SingleAxisTracker()
data = Plib.singleaxis(apparent_zenith=zenith['SolarZenithAngle'],apparent_azimuth=azimuth['Azimuth'])
print(data)`
**Expected behavior**
Should have got the tracking values

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions:**
 - ``pvlib.__version__``: 0.7.1
 - ``pandas.__version__``: 1.0.1
 - python: 3.8.1

**Additional context**
Similar to issue #622 Didnt find a solution there or anywhere else

cwhanse, 2020-03-10:
Did you not get output from the single axis tracking function?  

If the input data include sun position behind any possible rotation of the tracking plane, the code will emit warnings from numpy functions, see discussion [here](https://groups.google.com/d/msg/pvlib-python/c8744ODDgtA/-nHNRoVTGQAJ)

HrushikeshBodas, 2020-03-14:
I got a matrix in which all values are Nan. The values for tracking plane lie between -90 to +90 right?

wholmgren, 2020-03-14:
@HrushikeshBodas we'll need more information to determine if this is a bug. Can you post a small subset of the contents of azimuth and zenith for which you think the function should return non-nan values? For example, `azimuth.iloc[0:2]` and `zenith.iloc[0:2]`. Thanks.

HrushikeshBodas, 2020-03-15:
@wholmgren I ran the function again with different values, and It give me tracking values which were fairly consistent. So I dont think this is a bug, just bad input data in this case

mikofski, 2020-03-16:
@HrushikeshBodas, we frequently see issues when the datetime series is not timezone aware. Can you make sure that the timezone is correct? You can use either the [`tz` argument in `pandas.DatetimeIndex`](https://pandas.pydata.org/pandas-docs/stable/reference/api/pandas.DatetimeIndex.html) or the [`.tz_localize` method](https://pandas.pydata.org/pandas-docs/stable/reference/api/pandas.DatetimeIndex.tz_localize.html#pandas.DatetimeIndex.tz_localize) to convert a naive datetime series to timezone aware. A quick way to tell is to check the solar position, if the solar zenith is less than 90&deg; during the day and a minimum near noon.

**If you find you have now resolved this issue, perhaps you can close it? Thanks!**
[image]

Also I encourage you and others to read through and/or post on the [pvlib Google Group](https://groups.google.com/forum/#!forum/pvlib-python) or with the [`[pvlib]` tag on StackOverflow](https://stackoverflow.com/tags/pvlib) before posting an issue. For more information, please see [the documentation section on getting help](https://pvlib-python.readthedocs.io/en/latest/package_overview.html#getting-support).

Cheers!

HrushikeshBodas, 2020-03-17:
Sure Thanks a lot for your help!

closed 2020-03-17 by HrushikeshBodas