pvlib/pvlib-python #935: move kimber and HSU to soiling.py and remove losses.py

mikofski, 2020-03-17:
**Is your feature request related to a problem? Please describe.**
see #931 - the question is where to put different losses like snow, soiling, but not all effects are considered losses by all, so where would transformers go? or curtailment? it's a slipper slope, where would inverter efficiency go? Everything in the loss tree would get jumbled together

**Describe the solution you'd like**
rename [`losses.py`](../blob/master/pvlib/losses.py) to `soiling.py`

**Describe alternatives you've considered**
put all losses of different types into `losses.py` except then some things that might not be considered losses will be confusing?

**Additional context**
see #931

cwhanse, 2020-03-17:
@mikofski by opening this issue, should we assume you are +1 to this [question](https://github.com/pvlib/pvlib-python/issues/891#issuecomment-600301656)?

mikofski, 2020-03-17:
Yes, I think it's a good idea, I believe that `losses.py` is likely to get quite large, and like someone else said, maybe it was Steve Ransom, there maybe many items in the loss tree that are not necessarily considered _losses_ by everyone.

However, it might make sense later to think of a subpackage to group/organize loss-tree items into. In SolarFarmer, they are collectively called _effects_ instead of _losses_. So losses are just negative effects, whereas a boost/gain is a positive effect. But, I don't think we have to make that decision at this time, b/c the `pvlib` package is perhaps not that crowded yet. I don't know how to decide when it becomes _too_ crowded, probably when people start complaining or there's some other issue.

mikofski, 2020-03-17:
closed in favor of #891 sorry for the clutter - I got confused :(

cwhanse, 2020-03-18:
actually I thought you were getting ahead - let's use this issue to move `losses.py` to `soiling.py`.

PR #937 (2020-04-07): rename losses.py to soiling.py
 - [x] Closes #935
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pu [... cut]
closed 2020-04-07 by CameronTStark