pvlib/pvlib-python #939: TEMPERATURE_MODEL_PARAMETERS is undocumented

mikofski, 2020-03-20:
**Is your feature request related to a problem? Please describe.**
`TEMPERATURE_MODEL_PARAMETERS` is mentioned several times in the docs, but it isn't published in the api

**Describe the solution you'd like**
add a docstring after the constant definition in [`temperature.py`](../blob/e526b55365ab0f4c473b40b24ae8a82c7e42f892/pvlib/temperature.py#L9) like this (exact wording tbd):
```python
TEMPERATURE_MODEL_PARAMETERS = {
    'sapm': {
        'open_rack_glass_glass': {'a': -3.47, 'b': -.0594, 'deltaT': 3},
        'close_mount_glass_glass': {'a': -2.98, 'b': -.0471, 'deltaT': 1},
        'open_rack_glass_polymer': {'a': -3.56, 'b': -.0750, 'deltaT': 3},
        'insulated_back_glass_polymer': {'a': -2.81, 'b': -.0455, 'deltaT': 0},
    },
    'pvsyst': {'freestanding': {'u_c': 29.0, 'u_v': 0},
               'insulated': {'u_c': 15.0, 'u_v': 0}}
}
"""Dictionary of temperature parameters organized by model.

There are keys for each model at the top level. Currently there are two models,
``sapm`` for the Sandia Array Performance Model, and ``pvsyst``. Each model has
a dictionary of configurations with a dictionary of the model parameters
associated with it. Retrieve a parameters by indexing the model and
configuration by name.

Example
-------
Retrieve the open rack glass-polymer configuration for SAPM::

    from pvlib.temperature import TEMPERATURE_MODEL_PARAMS
    temp_parmas = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_polymer']
"""
```

Then somewhere in the docs add an [`autodata` directive](https://www.sphinx-doc.org/es/master/usage/extensions/autodoc.html#directive-autodata) to publish the constant data to the api, like this:
```
Temperature Model Parameters
~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.. currentmodule:: pvlib.temperature
.. autodata:: TEMPERATURE_MODEL_PARAMETERS
   :annotation:
```
_note_: remove `:annotation:` to show the value of the dictionary

depending on where it goes, you might have to return the module namespace to whatever it was before this entry using something like this:

    .. currentmodule:: pvlib

to return it to `pvlib` for example if this was in `api.rst`

renders like this:
[image]

And now it's in the index
[image]

and can be referenced everywhere in the docs using:

    :data:`~pvlib.temperature.TEMPERATURE_MODEL_PARAMETERS`

Ideas for where to put it?
- in `api.rst in the subsection called "PV Temperature Models" part of "PV Modeling"
- in `pvsystem.rst` at the end after SAT, in a new section called "temperature"
- huh?

**Describe alternatives you've considered**
- instead of using a dictionary, use a constant class and then add it to the temperature section of `api.rst` after the Faiman model, but I tried this for a while and I couldn't do it easily, or it was too late

**Additional context**
see #938

wholmgren, 2020-03-21:
Thanks @mikofski for the thorough suggestion. +1 for documenting the dict, -1 for the class.

wholmgren, 2020-08-26:
@mikofski can you make a PR with this for 0.8 (next few weeks)? If not maybe @kanderso-nrel ?

PR #1036 (2020-09-02): [DOC] add TEMPERATURE_MODEL_PARAMETERS to api.rst

 - [x] Closes #939
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] ~Tests added~ no code, only documentation
 - [ ] Updates entries to [`docs/sphinx/source/api.rst`](https://github.com/pvlib/pvlib-python/blob/master/docs/sphinx/source/api.rst) for API changes.
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/master/docs/sphinx/source/whatsnew) for all changes. [... cut]
closed 2020-09-02 by wholmgren