pvlib/pvlib-python #940: BUG: microseconds treated as seconds in unused function spa.julian_day_dt

kanderso-nrel, 2020-03-21:
Microseconds aren't scaled before adding with other time components, meaning they get treated as seconds: [spa.julian_day_dt](https://github.com/pvlib/pvlib-python/blob/master/pvlib/spa.py#L433) 

```python
In [53]: pvlib.spa.julian_day_dt(2020, 1, 1, 0, 0, second=1, microsecond=0)
Out[53]: 2458849.500011574

# should return same as above but doesn't
In [54]: pvlib.spa.julian_day_dt(2020, 1, 1, 0, 0, second=0, microsecond=1e6)
Out[54]: 2458861.074074074

# shows us and s treated the same
In [55]: pvlib.spa.julian_day_dt(2020, 1, 1, 0, 0, second=1e6, microsecond=0)
Out[55]: 2458861.074074074
```

Note -- I don't think this function actually gets called anywhere, so I don't think it is causing any issues with solar position calcs or anything.  I only noticed it because I was reading through spa.py's source code.

ericf900, 2020-03-23:
Hey @kanderso-nrel 
Just opened a PR to fix this issue! It updates both the function and the test for spa.julian_day_dt. :pull:`942`

kanderso-nrel, 2020-04-01:
Closed by #942

closed 2020-04-01 by kanderso-nrel